Ultimaker/CuraEngine #339: Add support to convert each slice layer to SVG

Zythyr, 2016-05-11:
I would like to request adding support for converting each sliced layer to SVG. This would be very useful for DLP based 3D printers.

daid, 2016-05-11:
It's not the goal of the CuraEngine to support other printing methods then FDM.

DLP should have it's own proper toolchain. (And DLP is 100x easier to do then FDM)

closed 2016-05-11 by daid

BagelOrb, 2016-05-11:
If you give me a DLP printer then I will make cura work with it!
:D

Marviel, 2018-01-10:
It's so much easier, in fact, that CuraEngine basically already does all the stuff you'd need to do to enable this feature.

If there were a pull request submitted with this feature implemented well, what are the chances it would be pulled in?

ChrisTerBeke, 2018-01-10:
As Ultimaker does not have any DLP machines, I don't think we'll spend time reviewing/testing that, especially because the way that would have to communicate with the Cura front-end would be totally different (as it's designed for FDM at the moment), so it wouldn't be an end-user feature anyways (unless you feel comfortable with using the engine via the command line).

However you could totally use your own fork of CuraEngine to use the feature of course.

Marviel, 2018-01-11:
@ChrisTerBeke thanks for quick reply!

Makes sense, was just curious if that was in the cards. Was only thinking the command line element anyhow. Thanks!

BagelOrb, 2018-01-11:
I think exporting each layer to SVG could have some very nice debugging value for Ultimaker.

I would vote to review the feature when a PR arives, though we shouldn't spend too much time debugging it.

There is already a class for exporting to SVG in the engine, so implementing the feature shouldn't be too difficult.

Make sure there's a setting or a command line argument to enable this feature, so that normal processing is still the default if you don't fill in that setting/argument.