Ultimaker/CuraEngine #341: vertex meld distance

sajjadul-wematter, 2016-05-18:
Hello forum,

I am going through the CURA source and right now I am at the mesh.cpp. You have a static function named pointHash(...) that doing some conversion  - probably from mm to integer steps. The function is defined without any documentation and it would be great if some explanation is provided behind its usage and definition.

Thanks

daid, 2016-05-18:
It makes a hash from a 3D point? To be used in a hashmap later on. Not really rocket science...

BagelOrb, 2016-05-18:
It returns a hash for the location, but first divides by the vertex_meld_distance, so that any point within a box of vertex_meld_distance by vertex_meld_distance would get mapped to the same hash.

closed 2016-05-18 by BagelOrb