Ultimaker/CuraEngine #344: Machine Configurations should be able to inherit absolute paths

jrussellsmyth, 2016-05-22:
The current implementation of SettingRegistry::loadJSONsettings assumes that "inherits" member specifies a path relative to the inheriting file. This does not allow, for example, a private configuration file to inherit from a public/shared profile

BagelOrb, 2016-05-22:
Indeed; we are currently reworking how settings work and how they are saved to file.

The branch settings_rework contains new code for the new .def.json files.
The new way to do it (currently) is by loading the json files 'by hand', e.g. `CuraEngine -j ..../fdmPrinter.def.json -j ..../yourPrinter.def.json`
For some machines this can become quite combersome: e.g
`CuraEngine -j ..../fdmPrinter.def.json -j ..../ultimaker.def.json -j ..../ultimaker2.def.json -j ..../ultimaker2_plus.def.json -j ..../ultimaker2_extended_plus.def.json`

We are still in the process of reworking the way settings are handled, but it seems to me that above strategy is the only viable way out of the problematic situation we are in now - as you rightfully pointed out.

BagelOrb, 2016-05-22:
Note that since the same machine definition might be on a different location on each computer, the files cannot refer to an absolute path.

Another way to deal with the problem is to have two standard locations: the internal location and the location for user supplied definition files; these default locations could then be recorded in a number of ways, e.g. as environment variables.

Come to think of it, that would be a preferable solution, don't you think?

jrussellsmyth, 2016-05-23:
Having 2 (or more) specified locations for files, and searching within those based on a file spec could be workable - which locations could be part of the installation configuration. This would work similar to a Java classpath (giving path roots and all paths then can be searched in each root. Ex:

given
resource_root = ${install-root}/resources,${home}/.cura/resources

Then inherit = machines/fdmprinter.json would be the first file found in 
${install-root}/resources/machines/fdmprinter.json
${home}/.cura/resources/machines/fdmprinter.json

This would work very well

BagelOrb, 2016-05-25:
See https://github.com/Ultimaker/CuraEngine/tree/settings_rework#running

closed 2016-05-25 by BagelOrb

jrussellsmyth, 2016-06-06:
This link is 404

BagelOrb, 2016-06-06:
https://github.com/Ultimaker/CuraEngine/#running