Ultimaker/CuraEngine #345: parent child relationships within the closed loops after the slicing operation

sajjadul-wematter, 2016-05-23:
Hello 

I am looking for the portion within the source code where the the parent-child hierarchy is defined within the list of the closed loops (or list of polygons ) after the slicing operation and before the tool-path generation. 

Thanks

BagelOrb, 2016-05-23:
It isn't. That information is never retained from the clipper library. It's
only used when performing polygon::splitIntoParts...

The information is present in a clipper::PolyTree but that type isn't
really used in the engine.

sajjadul-wematter, 2016-05-23:
Interesting! If it is not then, how do you differ the inner loops from the outer one. An outer closed loop may contain several holes within it and the region in between the outer loop and inner loop has to be filled with material. How does it work then during tool-path generation if the relationship is not well-defined ?

BagelOrb, 2016-05-23:
I think the loops are saved in a vector, and added consecutively by
performing offsets to the last wall.

The gaps between walls are currently not filled (anymore). We could
generate these gaps during the generation of the walls.

sajjadul-wematter, 2016-05-23:
I am not sure if I managed to explain the issue clearly. Here goes a link to an image that may describes my query. 

[partition of layer](http://imgur.com/so8C2Gl)

As you can see in the image there are outer loops and several nested loops. And most importantly the filled region. My question is that which class is doing the job of separating several nested loops and fills those region based on the condition of intersection into and intersection out of the contour by the tool-path .

BagelOrb, 2016-05-23:
You should take a look at sliceDataStorage.h that's where all the data is
stored.
FffPolyginProcessor fills those days structures mostly during
processInsetsAndInfill or something. (I'm currently unable to see the
source)

That's where the outlines of a model are used to generate the outer walls
(nested loops) and skin(dense top/bottom infill) as well as the sparse
infill area.

daid, 2016-05-24:
Clipper can do this all. We're using the "simple" interface of clipper mostly, which defines holes are counter clockwise and outlines as clockwise (or the other way around). Clipper can return a tree structure to define outlines with holes in it. But we're only using this at a few specific spots.

sajjadul-wematter, 2016-05-24:
I do undestand the fact that not all part of clipper is used to fulfill the requirements of Cura engine - I am interested about those "specific spots" where you are using clipper at cura to identify and differentiate between the outer and inner polygons.

sajjadul-wematter, 2016-05-24:
The main reason I want to get into that much of details is because I have developed a tool-path based on fractal (Hilbert Space Filling Curve) and I want to attach it as a infill tool-path for Cura. I need to know which part of the source I need to study to make the necessary adjustments within my code so that it fits in Cura.

[Fractal Tool-Path](http://imgur.com/qH8qswy)

daid, 2016-05-24:
Well, a lot of code does not differentiate. As the polygons follow the "even/odd" rule.
For example, the line infill algorithm does not look at "inner" or "outer" polygons, it just look at any line crossing:
https://github.com/Ultimaker/CuraEngine/blob/master/src/infill.cpp#L161

BagelOrb, 2016-05-25:
https://github.com/Ultimaker/CuraEngine/blob/master/src/utils/polygon.cpp#L245 will convert a list of polygons into a list of PolygonsPart

In your example a PolygonsPart is `{a, c, d}` (the outer polygon always first)
The result of the function will then be `{ {a, c, d}, {e}, {f, g, h}, {j}, {b, i} }`
The fact that `{f, g, h}` is inside `{a, c, d}` is not recorded whatsoever.

sajjadul-wematter, 2016-05-28:
Thanks for the hint. Inside the polygon.h you have declaration for the classses PolygonRef, Polygons, PolygonsPart . None of these classes validate the fact that the path that you store inside them do indeed is a closed path. Is it strongly assumed from the very beginning or it is checked after the slicing operation ?

BagelOrb, 2016-05-28:
All polygons are assumed to be closed paths. At some placed we indeed use open polylines, but it should be very well documented that those places.

We don't verify that the paths are closed - they are used as closed paths. A closed path `{p0,p1,p2}` is a polygon with line segments `p0-p1`, `p1-p2` and `p2-p0`. Such a polygon is not stored as `{p0,p1,p2,p0}`...

Furthermore, at quite some places we use a polygon with only two points to signify a single line segment.

sajjadul-wematter, 2016-05-28:
The Subject polygon may or may not be a closed polygon , but the clip polygon must be a closed polygon. Is not that ? And you mentioned that you are not validating the fact that the clip polygon is indeed a closed path. Is it because it will be closed by default after the slicing and extensive stitching operation ?

BagelOrb, 2016-05-28:
The slicing stage only provides closed polygons indeed. All line segments
which are not used in finding closed polygons are stored to use for the
mesh surface mode feature, which is generally turned off.

It doesn't make sense to talk about checking whether a path is closed or
not.
Whether a path is closed is not something to be checked, but information to
be provided to clipper. The difference between using a vector of points as
an open or closed path is simply that clipper fills in the last line
segment from the last to the first point in the vector or not.

sajjadul-wematter, 2016-06-07:
While skimming through the function Polygons::splitIntoParts(....) , It is noticed that another private is further processing resulting "resultPolyTree" to organize the contour according to the parent-child relationship. Is there any precondition to call the function Polygons::splitIntoParts(....) - like the orientation of the closed paths. (for example ; The orientations of closed paths must be consistent such that outer polygons share the same orientation, and any holes have the opposite orientation (ie non-zero filling). Open paths must be oriented with closed outer polygons. ).

BagelOrb, 2016-06-07:
I think that precondition holds, but I'm not sure.

BagelOrb, 2016-06-07:
Take a look at http://www.angusj.com/delphi/clipper.php

sajjadul-wematter, 2016-06-07:
Where do you mention the clipping window in the function Polygons::splitIntoParts(....) ? I can only see the list of  subject paths.

BagelOrb, 2016-06-07:
What do you mean by a 'clipping window'?

sajjadul-wematter, 2016-06-07:
I am extracting the contents from clipper - "Clipping: commonly refers to the process of cutting away from a set of 2-dimensional geometric shapes those parts that are outside a rectangular 'clipping' window. This can be achieved by intersecting subject paths (lines and polygons) with a clipping rectangle. In a more general sense, the clipping window need not be rectangular but can be any type of polygon, even multiple polygons. Also, while clipping typically refers to an intersection operation, in this documentation it will refer to any one of the four boolean operations (intersection, union, difference and exclusive-or). "

I believe that you are using clipper.

BagelOrb, 2016-06-07:
polygon.difference(Polygons& other) subtracts the other polygon from this
polygon.

Ghostkeeper, 2016-06-08:
Clipping is indeed a term for cutting away geometry, but "Clipper" or "ClipperLib" is also the name of a library that does geometric computations. CuraEngine uses that library to do its offsets and compute binary operations on polygons.

`meshfix_union_all` and `meshfix_union_all_remove_holes` are the identifiers of two settings. For reference, take a look at https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json#L2042 . These two settings are meant to repair meshes that aren't completely simple, e.g. with superfluous or missing faces.

`Polygon::splitIntoParts` doesn't necessarily hold with one specific convention for the ordering of vertices or parts in a polygon, but assumes that ClipperLib is consistent in the ordering it gives back to CuraEngine and the ordering it expects from CuraEngine.

closed 2016-10-10 by BagelOrb