Ultimaker/CuraEngine #346: 2.1.1: Errant Extrusion Move

CCS86, 2016-05-27:
Hi guys,

I thought I had found a bug in in Repetier's visualization, but on closer inspection, I think it might be a Cura issue.

Check out this sequence:

[image]

We finish the outer perimeter with a coast move, comb into position for the other outer wall, then just sit there and extrude.

I calculated the extrusion rate for some of those "fat" print moves, and compared to the normal ones, and found them the same. I think the fatness is just Repetier's way of distributing out the extra extrudate.

I printed this part, and found a number of this bizarre defects in the surface, which is why I went hunting.

Model:
[Cover2.stl.txt](https://github.com/Ultimaker/CuraEngine/files/286971/Cover2.stl.txt)

Gcode file:
[cover2.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/286967/cover2.gcode.txt)

BagelOrb, 2016-05-27:
Coasting is implemented quite cuboptimally atm.

Coasting is like controlled oozing. We need to catch up the oozed material, otherwise we won't extrude enough material. We currently catch up by forwarding the material just like an unretract/prime move, but it would be better to catch up by gradually starting to move during this priming.

I implemented coasting by thinking it was to do with molten plastic left behind in the extruder after a retract, but what people want coasting for is to deal with bowden pressure.

The perfect coasting algorithm would result in something like this:
1) At the end of an extrusion move we stop feeding and let the bowden pressure bleed out and gradually recude the printing speed accordingly.
2) At the start of an extrusion move we gradually build up the bowden pressure by feeding at a higher rate and we gradually increase the sprint speed from zero accordingly.

This is quite hard to implement, because there are a lot of factors to take into account (maximal feedrate, relation between bowden pressure and feedrate, etc.)

CCS86, 2016-05-27:
Ahh, that makes sense. 

It seems like a very small move, so I'm not sure it was to blame for my print defects or not. There seems to be a little glitch in the Repetier visualizer, because on neighbor layers, the same move is there, without the fat bead of filament.

closed 2016-06-10 by CCS86

michalka, 2017-02-15:
This is causing over extrusion at the start of my outside walls.  If I have zseam aligned, it causes a larger zseam than when coasting is disabled.  If I have randomized zseam this causes little bumps all over my print.  I really don't think it's necessary to re-pressurize after a coast, but if it is, this definitely isn't the way to do it.

BagelOrb, 2017-02-20:
We don really use coasting because of this reason.

Perhaps you should just turn coasting off.

Weŕe not really focussed on coasting atm, but if you like you could try to fix the problem yourself and submit a pull request...