Ultimaker/CuraEngine #348: Make Coasting Moves G1?

CCS86, 2016-06-06:
Hi guys,

Am I correct that only G1 moves are modified with the feed override knob? If that's true, it would seem like the G0 coast implementation would cause a speed difference compared to normal printing moves, if the override was not 100%.

BagelOrb, 2016-06-06:
Feed override knob??

Both the firmware and the slicing engine treat coasting moves wrong. Indeed
the engine should output G1s, but I think Marlin distinguishes between
travel and extrude moves based on the difference in E value.

CCS86, 2016-06-06:
Whatever you want to call it. The override knob. Know what I mean?

I believe it leverages M220.

BagelOrb, 2016-06-06:
I genuinely don't know which knob you are referring to, but M220 works on
travel and extrusion moves equally.

CCS86, 2016-06-06:
[image]

CCS86, 2016-06-06:
I know the override functionality isn't enabled on every machine, but I believe it is on all Ultimakers.

daid, 2016-06-07:
However, the firmware code does not distinguish between different moves with G0 and G1. They are identical for the firmware.

CCS86, 2016-06-10:
I see, I see. I guess that means this is a non-issue then :)

closed 2016-06-10 by CCS86