Ultimaker/CuraEngine #352: Request: Minimum Length for Brim Segment

CCS86, 2016-06-11:
Hi guys,

Would it be possible to put a limitation on the brim generation? Maybe either a minimum brim segment length, or pocket size.

The issue I have, is that it tries to add brim into a very small void inside the part:

[image]

DDDirk, 2016-06-12:
I also sometimes have this problem, and also when there is a hole in the bottom (eg to fit a bolt through). If the purpose is to prevent warping, I assume that is less likely to happen when it is an 'inner edge', such as in the latter case, and/or when it's a small space.
Usually, it is not a big problem, but I can imagine in some cases it can be difficult to remove.
But maybe Cura should not guess at what users will want in general terms, but give an option to remove certain sections of the brim.

BagelOrb, 2016-06-13:
I will introduce a setting which will only put brim on the outside; is that a good solution?

CCS86, 2016-06-13:
I think that could help.

But I can't help but think about a print with a very large inner cavity
that could use some brim as well.

What about a minimum inside cavity size? Then you could set that value very
high to exclude all insides, or tune it to suit.

Ghostkeeper, 2016-06-14:
Never hurts to ask ;) But clicking on pieces of the layer view to remove it probably won't be implemented anytime soon.

closed 2016-10-10 by BagelOrb