Ultimaker/CuraEngine #355: Where can I get a list of command line commands currently supported and also planned near future

3JS2016, 2016-06-17:
Where can I get a list of command line commands currently supported and also planned for the near future.  Is there any list/plans of working on future features?

Ghostkeeper, 2016-06-17:
You can get a list of command line commands when you call CuraEngine from the command line without any arguments or with unknown arguments (such as `--help`). For reference, the instructions you get are implemented here: https://github.com/Ultimaker/CuraEngine/blob/master/src/main.cpp#L26

We currently have no plans of extending the command line interface of CuraEngine, apart from that we're not entirely happy with how it needs to find the JSON files for machine definitions. That still needs some thought. Cura's front-end doesn't use the command line arguments to slice, so it has no real priority. Cura's front-end makes a socket connection to the engine by starting it with `CuraEngine connect -p <port>` and then sending the mesh and settings via Protobuf-packages.

BagelOrb, 2016-06-17:
For a list of all settings, look at the [standard list of settings](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json), the [settings which only apply to a nozzle etc.](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmextruder.def.json) and [settings only used when using CuraEngine as command line tool](https://github.com/Ultimaker/CuraEngine/blob/master/resources/command_line_settings.json)
The last one is outdated..

closed 2016-06-20 by BagelOrb