Ultimaker/CuraEngine #356: cura与curaEngine的如何通讯问题

Twosilly, 2016-06-17:
English：
sorry!
My English is not very good, then these I am through the translation software translation!
I really bother you!
Hello! I am a student! Because I love 3D so it has been in print!! I want to own a slice of rewriting! But I technology is not very good! I couldn't find Cura how python with Engine or Python communication! What is passing a parameter to Engine let him to complete the perfect slice! I know that engine will accept the command line parameters -h help! Version information -v engine! -m is a rectangular hot bed is information, you can also move the print head after the clip regions are removed as -c -s I only know that they are delivering information but how to pass the set! From what I have been unable to accept! -o is a named.Gocde and you want to slice model!
If you are free, I really trouble you to help me explain the principles of -c and -s!

你好！我是一名学生！因为很喜欢3d打印！所以就一直在研究！我想重写一个自己的切片软件！不过我的技术不是很好！我始终找不到cura的python是如何跟Engine通讯的！或者说python是传递了些什么参数给Engine让他来完成完美的切片的！我知道 engine会接受命令行参数 -h帮助信息！-v engine的版本信息！-m是热床的矩形信息，就是除去夹子后打印头还可以移动的区域，至于-c -s 我只知道他们是传递设置信息的！可是如何传传什么我一直找不到！-o 是接受一个 命名的.gocde和你要切片的模型！
如果您有空的话，真心的麻烦您给帮我解释一下 -c和-s的原理！谢谢！

Ghostkeeper, 2016-06-17:
There is no `-m`.
There is no `-c`.

`-s` changes a setting. You'll say `-s name=value`. The name comes from this list: https://github.com/Ultimaker/Cura/blob/2.1/resources/machines/fdmprinter.json

Example: `CuraEngine.exe -j resources/machines/fdmprinter.json -s layer_height=0.2 -l mesh.stl`

BagelOrb, 2016-06-17:
Which version of Cura are you running?

Twosilly, 2016-06-20:
@Ghostkeeper 
First of all thank you very much! 
I will go back and study it carefully!
Thank you again for your answer!

I use the engine is a version of 15.04.6! So there is no JSON module!
So how am I going to keep it in communication?

Thank you! 
chinese:

首先非常感谢你！
我会回去仔细研究它！
再次感谢你的回答！

我使用的引擎是一个版本的15.04.6！所以没有JSON模块！

那么我如何保持它在沟通呢？

谢谢您！   给你添麻烦了！

Twosilly, 2016-06-20:
@BagelOrb 
First of all, thank you very much!
I'll go back and study it carefully!
Thank you again for your answer!

I used the engine to be a version of 15.04.6!

So do you have any good suggestions?
Thank you!
chinese:
首先非常感谢你！
我会回去仔细研究它！
再次感谢你的回答！

我使用的引擎是一个版本的15.04.6！
所以 你有什么好的建议吗？

谢谢您！

Ghostkeeper, 2016-06-20:
That's in here: https://github.com/daid/Cura/blob/SteamEngine/Cura/util/sliceEngine.py#L176
It also connects via a socket like we do now, but doesn't use Protobuf yet. But I assume that there was also a CLI.

BagelOrb, 2016-06-20:
Here's an old example run I used to run:
./CuraEngine -v -s initialSpeedupLayers=4 -s minimalFeedrate=10 -s preSwitchExtruderCode= -s supportExtrusionWidth=600 -s supportXYDistance=700 -s insetXSpeed=50 -s retractionZHop=0 -s extruderOffset[3].X=0 -s extruderOffset[3].Y=0 -s gcodeFlavor=1 -s postSwitchExtruderCode= -s retractionSpeed=40 -s retractionPrimeSpeed=30 -s filamentFlow=100 -s infillOverlap=15 -s inset0Speed=50 -s coolHeadLift=0 -s extrusionWidth=400 -s upSkinCount=6 -s initialLayerSpeed=20 -s minimalLayerTimeFanSpeedMin=2000 -s minimalLayerTime=5 -s infillSpeed=50 -s supportExtruder=-1 -s fanSpeedMax=100 -s supportType=1 -s enableCombing=1 -s fanSpeedMin=100 -s supportZDistance=150 -s supportOnBuildplateOnly=0 -s filamentDiameter=1128 -s initialLayerThickness=300 -s supportAngle=60 -s fanFullOnLayerNr=2 -s extruderOffset[1].X=18000 -s extruderOffset[1].Y=0 -s layerThickness=100 -s endCode=M25 -s minimalExtrusionBeforeRetraction=20 -s retractionMinimalDistance=1500 -s skirtMinLength=150000 -s retractionAmount=4500 -s retractionPrimeAmount=0 -s skinSpeed=50 -s skirtLineCount=1 -s startCode= -s skirtDistance=0 -s extruderOffset[2].Y=0 -s extruderOffset[2].X=0 -s printSpeed=50 -s fixHorrible=1 -s layer0extrusionWidth=400 -s moveSpeed=150 -s supportLineDistance=2666 -s retractionAmountExtruderSwitch=16500 -s sparseInfillLineDistance=2000 -s insetCount=2 -s downSkinCount=6 -s multiVolumeOverlap=150 -s posx=115000 -s posy=112500 -o "test.gcode" "../3D_models/Cura_test.stl"

Twosilly, 2016-06-22:
@Ghostkeeper 
Thank you very much for your answer! I'm almost sure how to do it!

（I really appreciate you can put up with my Chinese English to help me solve the problem!
）
真的非常感谢你的回答！我已近大概清楚该怎么去研究了！万分感谢！
真的非常感谢你可以忍受我的中式英语帮我解决问题！再次表示感谢！

Twosilly, 2016-06-22:
@BagelOrb 
Thank you very much for your answer! I'm almost sure how to do it!
Your answer saves me most of the time!

（I really appreciate you can put up with my Chinese English to help me solve the problem!
）
真的非常感谢你的回答！我已近大概清楚该怎么去研究了！万分感谢！
你的回答节省了我大部分的时间！谢谢！
真的非常感谢你可以忍受我的中式英语帮我解决问题！再次表示感谢！

closed 2016-09-04 by Ghostkeeper