Ultimaker/CuraEngine #357: 2.1.2 Odd moves

CCS86, 2016-06-20:
Hi guys,

These sequence seemed very out of place when watching the print. Do you know why it was generated?

[image]

[model.stl.txt](https://github.com/Ultimaker/CuraEngine/files/322597/model.stl.txt)

[gcode.txt](https://github.com/Ultimaker/CuraEngine/files/322596/gcode.txt)

BagelOrb, 2016-06-20:
I don't know. Did you set infill to 100%?

It's better to set the top bottom thickness to 9999 instead.

CCS86, 2016-06-21:
Yes, it is 100% lines. 

I don't like using bottom layers, because it used to disable infill wipe, and now maybe combing?

BagelOrb, 2016-06-21:
If you set combing to No Skin then it won't do combing in 100% mode.

Why would you want infill wipe when you only have skin?

Why do you think the order of lines is weird? It first does the infill and
then the top bottom skin. This is to minimise the number of changes in
nozzle pressure.

CCS86, 2016-06-21:
It's not the order of the lines I find odd; it's that they exist at all. That space should easily fill without gap filling, and did on the prior layers. Have a look at layers 1 and 2, to see what I mean.

I'm not sure I understand the question about infill wipe. I want infill wipe everywhere, because it reinforces the bond between infill and perimeters. I don't know what you mean by "only have skin".

Ghostkeeper, 2016-06-21:
When infill is set to 100%, by default it makes the number of skin layers 999999 (because that tends to work better). He's therefore assuming that you only have skin and no actual infill.

CCS86, 2016-06-21:
Do "top" and "bottom" layers both classify as skin?

I feel like in one of the recent releases, with 100% infill set, I could see the difference between bottom layers and normal infill, by the presence of infill wipe.

Ghostkeeper, 2016-06-22:
Yes, they are both skin.

CCS86, 2016-06-23:
Do you guys agree, that this sequence of moves does not make sense?

BagelOrb, 2016-06-23:
No, to me or makes perfect sense. Think about what happens when you make the infill 95%. Then the lines won't line up any more. The fact that the infill and the skin line up is just a coincidence which makes it *seem* like it's doing word detours.

The thing is that you'd want to do the infill lines and skin lines separately because they might have different speeds and changing speed a lot of times affects print quality.

You want to first print all infill lines and then all skin lines.

closed 2016-06-23 by BagelOrb

Ghostkeeper, 2016-06-23:
To me it still doesn't make sense :disappointed:

If skin layers is set to 999999, there should be no infill lines in the picture. Nothing is done separately.

CCS86, 2016-06-24:
I don't understand your reasoning @BagelOrb 

There are three, discrete areas that are being filled completely out of order with the rest of the infill. They just so happen to be symmetrically placed relative to the part geometry. I can't see the value in taking what should rightly be continuous infill lines throughout the entire infill region, and breaking it up like this.

I have noticed this sort of behavior before, but never isolated it. It definitely has the capacity to lower print quality.

BagelOrb, 2016-06-28:
As I understood it you set the infill to 100% and then changed the top and
bottom skin back to the normal amount of layers.

I simply suggest that you leave the bottom skin count at 99999 if you don't
like this outcome.

The reason that it happens is that the infill areas are printed separately
from the top/bottom skin. The fact that the infill line distance is the
same as the skin line distance doesn't change that.