Ultimaker/CuraEngine #358: 2.1.2: Infill Type Erratic

CCS86, 2016-06-21:
I am seeing some erratic infill patterns. In this case, I selected concentric infill, but the issue is similar with lines, because the wall line count should dominate the infill on these loops. One looks perfect, the other two are consistently ignoring the 3 wall line count, and using line infill. All three features are identical.:

[image]

On an adjacent layer, it is even more strange, making all these little C's:

[image]

Other layers, more uniformly use lines infill:

[image]

[gcode.txt](https://github.com/Ultimaker/CuraEngine/files/324793/gcode.txt)
[model.stl.txt](https://github.com/Ultimaker/CuraEngine/files/324794/model.stl.txt)

BagelOrb, 2016-06-21:
If you don't want this, turn off Fill Perimeter Gaps or Remove Overlapping
Inner Wall Parts.

These two options will be gone in 2.2. instead overlapping walls will
extruder less. However, the gaps between walls can still occur without
being filled, when the walls are thinner than the nozzle size.

CCS86, 2016-06-21:
You were definitely right that "remove overlapping inner wall parts" was causing this behavior. I didn't realize that it disappeared from my visible settings, but was still on.

Without it, I see some other questionable things. What I would expect, is to lay in as many full width beads, as can fit, and then stop with anything left under the line width, leaving it for the gap filling algorithm. But, on some layers it appears that two full extrusion beads are nearly completely on top of one another:

[image]

While others are left with very large gaps:

[image]

Does the engine struggle with asymmetry in the inner loops? (can only fit one, but it wants one from each direction)

CCS86, 2016-06-21:
Hmm, just noticed that my setting for gap filling is nowhere to be found:

[image]

Ghostkeeper, 2016-06-21:
This setting was removed for 2.2 and replaced by a much simpler and more effective method: Reducing the line width until it fits. So we are not going to allow tuning this setting for the remainder of 2.1.

CCS86, 2016-06-21:
That is awesome to hear! I have always wished that it could  lay as many full width lines, as would fit, then finish with a partial width line.

BagelOrb, 2016-06-22:
Fill Gaps is only visible when remove overlapping inner wall parts is
enabled.

The new feature will still leave gaps such in your last picture, because it
has to perform the polygon offset on both walls the same time.

CCS86, 2016-06-23:
Are you saying that that in 2.2, the last picture would represent a "finished" layer?

BagelOrb, 2016-06-23:
Yes. This has always been a problem :S

In the old cura thin pieces could get overextrusion or underextrusion. At
least the first problem is solved..

CCS86, 2016-06-24:
What are the limitations that keep the new feature, "Reducing the line width until it fits", from working on this type of gap?

BagelOrb, 2016-06-24:
The problem is to detect when this is happening.

That detection algorithm will closely resemble wall overlap computation,
but instead of reducing the flow of particular line segments, new line
segments need to be introduced. This could become quite difficult when you
consider that the gaps between the walls can form a network rather than a
single line.

CCS86, 2016-07-01:
It still seems strange to me, that on a perfectly symmetric part, some portions are filled perfectly, and other identical portions are left blank. Is this rounding error?

Ghostkeeper, 2016-07-03:
Unless you've selected concentric infill, the position of the infill lines are computed to be at regular interval from the origin. The infill could be different if their distance to the origin is different.

If it's concentric infill like above, the infill should be more or less the same, bar rounding errors and the order in which it draws the infill.