Ultimaker/CuraEngine #360: infill_spare_density not change print time

roboticsexpert, 2016-06-23:
hi
i want generate gcode with some infill_spare_density for optimize price but when i change it , time of printing not change ....

 /home/3dfast/CuraEngine slice -s layer_height=0.1 -s infill_sparse_density=100 -j /home/3dfast/src/Cura/resources/machines/fdmprinter.json -o /home/3dfast/stls/12/6.gcode -e0 -l /home/3dfast/stls/12/file.stl -v

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Warning: setting an unregistered setting layer_height
Warning: setting an unregistered setting infill_sparse_density
WARNING: Loading polygon setting machine_disallowed_areas not implemented...
WARNING: Unrecognized data type in JSON: machine_platform_offset has type array
WARNING: Loading polygon setting machine_head_polygon not implemented...
WARNING: Loading polygon setting machine_head_with_fans_polygon not implemented...
Unregistered setting mesh_position_z
Unregistered setting mesh_position_y
Unregistered setting mesh_position_x
Unregistered setting center_object
Loaded from disk in 0.028s
Unregistered setting extruder_nr
 -s machine_depth="100" -s layer_height="0.1" -s infill_sparse_density="100" -s machine_extruder_count="1" -s machine_width="100" -s machine_center_is_zero="false" -g -e0 -e0 -l "0"
Unregistered setting adhesion_extruder_nr
Unregistered setting support_infill_extruder_nr
Unregistered setting support_roof_extruder_nr
Progress: start accomplished in 0.000s
Starting slice...
Slicing model...
Layer count: 98
Progress: slice accomplished in 0.103s
Starting layerparts...
Unregistered setting multiple_mesh_overlap
Progress: layerparts accomplished in 0.023s
Starting inset+skin...
Progress: inset+skin accomplished in 0.453s
Starting support...
Unregistered setting ooze_shield_enabled
Progress: support accomplished in 0.056s
Starting export...
Unregistered setting machine_use_extruder_offset_to_offset_coords
Unregistered setting machine_nozzle_offset_y
Unregistered setting machine_nozzle_offset_x
Unregistered setting machine_extruder_start_code
Unregistered setting machine_extruder_end_code
Unregistered setting switch_extruder_retraction_amount
Unregistered setting switch_extruder_retraction_speed
Unregistered setting switch_extruder_prime_speed
Unregistered setting prime_tower_flow
Unregistered setting prime_tower_line_width
Unregistered setting speed_prime_tower
Unregistered setting material_standby_temperature
Unregistered setting prime_tower_size
Unregistered setting support_extruder_nr_layer_0
Unregistered setting support_extruder_nr
Progress: export accomplished in 3.918s
Total time elapsed  4.57s.
Print time: 11080
Print time (readable): 3h 4m 40s
Filament: 14615

--------------------------
/home/3dfast# /home/3dfast/CuraEngine slice -s layer_height=0.1 -s infill_sparse_density=30 -j /home/3dfast/src/Cura/resources/machines/fdmprinter.json -o /home/3dfast/stls/12/6.gcode -e0 -l /home/3dfast/stls/12/file.stl -v

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Warning: setting an unregistered setting layer_height
Warning: setting an unregistered setting infill_sparse_density
WARNING: Loading polygon setting machine_disallowed_areas not implemented...
WARNING: Unrecognized data type in JSON: machine_platform_offset has type array
WARNING: Loading polygon setting machine_head_polygon not implemented...
WARNING: Loading polygon setting machine_head_with_fans_polygon not implemented...
Unregistered setting mesh_position_z
Unregistered setting mesh_position_y
Unregistered setting mesh_position_x
Unregistered setting center_object
Loaded from disk in 0.011s
Unregistered setting extruder_nr
 -s machine_depth="100" -s layer_height="0.1" -s infill_sparse_density="30" -s machine_extruder_count="1" -s machine_width="100" -s machine_center_is_zero="false" -g -e0 -e0 -l "0"
Unregistered setting adhesion_extruder_nr
Unregistered setting support_infill_extruder_nr
Unregistered setting support_roof_extruder_nr
Progress: start accomplished in 0.000s
Starting slice...
Slicing model...
Layer count: 98
Progress: slice accomplished in 0.058s
Starting layerparts...
Unregistered setting multiple_mesh_overlap
Progress: layerparts accomplished in 0.022s
Starting inset+skin...
Progress: inset+skin accomplished in 0.370s
Starting support...
Unregistered setting ooze_shield_enabled
Progress: support accomplished in 0.023s
Starting export...
Unregistered setting machine_use_extruder_offset_to_offset_coords
Unregistered setting machine_nozzle_offset_y
Unregistered setting machine_nozzle_offset_x
Unregistered setting machine_extruder_start_code
Unregistered setting machine_extruder_end_code
Unregistered setting switch_extruder_retraction_amount
Unregistered setting switch_extruder_retraction_speed
Unregistered setting switch_extruder_prime_speed
Unregistered setting prime_tower_flow
Unregistered setting prime_tower_line_width
Unregistered setting speed_prime_tower
Unregistered setting material_standby_temperature
Unregistered setting prime_tower_size
Unregistered setting support_extruder_nr_layer_0
Unregistered setting support_extruder_nr
Progress: export accomplished in 3.290s
Total time elapsed  3.78s.
Print time: 11080
Print time (readable): 3h 4m 40s
Filament: 14615

BagelOrb, 2016-06-23:
When calling the engine directly you should pass the setting called
infill_line_distance. The frontend usually computes that from the density.
The density isn't actually used by the engine.

roboticsexpert, 2016-06-23:
thanks a lot ... my problem solved ...

closed 2016-06-23 by roboticsexpert