Ultimaker/CuraEngine #363: Infill Thickness =/= Layer Thickness Request

CCS86, 2016-06-26:
Hey Guys, I'm printing right now with layer thickness at 0.1mm and infill thickness at 0.2mm. It is working beautifully, thank you!

What do you think about adding a modifier setting that would allow non-perimeter skin, to be printed at the infill thickness?

BagelOrb, 2016-06-26:
It shouldnt be that hard architecturewise, but requires some 'manual
labour' ;)

However, it doesn't make a lot of sense because I think it would impact
surface quality. Also it introduces extra travel moves and print moves
because regions in one layer which aren't present on the next should still
be printed at normal layer height.

The infill thickness only merges the intersection of two consecutive layers
into a thicker layer and leaves the rest to be printed with normal layer
height.

CCS86, 2016-06-26:
I see. That makes sense. Certain areas like that have to be printed at the layer thickness, or they can't be filled.

I asked because on this print, it has a very wide base, which tapers quickly. I wanted a certain bottom thickness, before the 25% triangle infill started, but during those bottom layers, it took forever at the 0.1mm thickness. 

Maybe a simpler fix for this case could do the perimeters of the bottom layers at layer thickness, but the solid lines infill at the thicker setting.

BagelOrb, 2016-06-26:
If it's model specific I'd advise to use the tweak at z plugin and just
print a different layer height for the first couple layers.