Ultimaker/CuraEngine #365: how can estimate slicing time or get slicing percent ?

roboticsexpert, 2016-06-27:
i want to have slicing percent or slicing time to show user how much time he should wait for getting gcode ...

how can i do this ?

BagelOrb, 2016-06-27:
I suppose you are using command line slicing rather than connecting via
command socket?

Use the argument -p

roboticsexpert, 2016-06-27:
thanks a lot ... 
i didn't find good documentation for using socket . if you know good place to find that please introduce it .

closed 2016-06-27 by roboticsexpert

BagelOrb, 2016-06-28:
Indeed it's not that well documented, but you should have a look at the
.proto file, how the engine works with it in commandSocket.h and you could
have a look at how the frontend sends messages.

Ghostkeeper, 2016-06-28:
Basically, you're going to need to write a program that uses [libArcus](https://github.com/Ultimaker/libArcus) to listen on some port for CuraEngine connecting. Then you can call CuraEngine with `CuraEngine connect -p <port>`.

Cura's front-end does the same here: https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/CuraEngineBackend.py