Ultimaker/CuraEngine #373: python coded program integration possibility

billy0329, 2016-07-20:
Hi,
I would like to know if it's possible to integrate CuraEngine's source to a python program.
Will it work for a program designed for windows?
If I'm gonna used it from source, do I still need Protobuf and libArcus?

I'm actually programming a 3dprinting software based on Pronterface/Printrun, which is programmed in python. My program currently works on Windows. 
My program, uses Slic3r as its current slicer but I've discovered that Slic3r can't slice some faulty stl files and those with huge amount of meshes. 
I would like to use CuraEngine as an alternative.

Can you guide me please.
I've just downloaded the source and will start my investigation.
But it would be better if I can get some ideas from you guys.

Thanks in advance. :-)

BagelOrb, 2016-07-20:
Our Cura front end is written in python, but we use libArcus to communicate
with the engine.

Currently it's necessary to restart the engine each time you slice, because
there are some global variables which don't get reset. I would therefore
think it's a dangerous move to integrate CuraEngine's source to a python
program.

Maybe @nallath has something more to say about this. @nallath?

Ghostkeeper, 2016-07-20:
Take a look at this file, Billy: https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/CuraEngineBackend.py

It specifies how the interface of Cura (written in Python) communicates with CuraEngine. It should serve as a good starting point, even if you just copy that code into your project.

nallath, 2016-07-20:
I think your time would be better spent in adding plugins to cura that improve the USB controls. That way you won't have to re-invent the wheel and Cura as a whole gets better.

billy0329, 2016-07-24:
Thanks a lot for your answers.

@BagelOrb , thanks for the info.
How about if I called the engine's binary, using subprocess in python?
I downloaded Cura 15.04.6, opened the exe in cmd, then run the following.

CuraEngine.exe -h 

then get the following usage:

usage: CuraEngine [-h] [-v] [-m 3x3matrix] [-c <config file>] [-s <settingkey>=<value>] -o <output.gcode> <model.stl>

may I ask how to use the following options:
[-m 3x3matrix]
[-c <config file>]
[-s <settingkey>=<value>]

it is right to say that if I already used a config file as an option, I can ommit using the settingkey-value pair? 
How can I have the configfile?

@Ghostkeeper , thank you for the info.
I've done a quick look on the codes. It's quite complicated than I thought. 
I think this would require me to undertand the whole processes of Cura as a whole.

@nallath , I would love to do so if I have the time. 
But right now I can't because of my other priorities.

:-)

BagelOrb, 2016-07-25:
> usage: CuraEngine [-h] [-v] [-m 3x3matrix] [-c ] [-s =] -o

You shouldnt work with the old cura. You should download the latest: Cura
2.1

May I ask: why are you looking into this? What's your mission?

closed 2016-10-10 by BagelOrb

billy0329, 2016-10-19:
@BagelOrb I'm very sorry for replying just now, because of my tight schedule here.

I'm asking those things for the reason that I'm working for [Novio3d](http://www.novio3d.com/).
And my task is to integrate cura to our custom software.
I was able to follow through the codes of the legacy cura version 15.04.6 and was able to 
make the slicing work within our software.
As cura being open-source, the company would like to know if there's any restriction if we distribute our software integrated with cura. 
The company wanted to be legal so we would like to know.

I apologize if it's inappropriate to ask this but I would be honest that I'm new to programming and barely understand whatever is written in AGPL license agreement.

Thank you. :-)

Ghostkeeper, 2016-10-19:
By the AGPL license, you are allowed to distribute modified or unmodified versions of the software. However, if you make a piece of software that depends on an AGPL project, it must also be AGPL licensed. "Depends" here is defined as that it wouldn't function without the dependency.

IANAL, so please correct me :wink:

billy0329, 2016-10-21:
thanks for the explanation @Ghostkeeper :-)

BagelOrb, 2016-10-28:
Please note that when you integrate Cura into your software you have to update your software every time Cura releases a new version or you accept getting behind.

You could also make it possible to select your machine from within Cura by submitting pull requests. That way you are asured that your printers are always using the latest version of Cura and have the benefit of all the advances Cura has made.

What I'm basically saying is that together we can achieve much more than when we are working on separate forks.