Ultimaker/CuraEngine #375: [patch] X3D support for CuraEngine

sevaa, 2016-08-02:
I've implemented support for the X3D scene format for CuraEngine.

X3D is an XML-based, open, royalty free format for describing 3D scenes. It's based on ideas from legacy VRML, but it's considerably larger. I've implemented an X3D parser that reads almost all mesh geometries in the standard, and feeds them into a Cura::Mesh object as triangles, not unlike the way STL parser does it.

Would the CuraEngine developer community be interested in such a contribution? If so, how do I go about integrating it into the codebase? It's two new files, a dependency (libxml2), and some lines in loadMeshIntoMeshGroup() in MeshGroup.cpp (it goes, basically: "if extension is STL, invoke the STL parser. Else if extension is X3D, invoke the X3D parser").

I didn't feel like implementing an XML parser from scratch, and integrated with libxml2 instead. If you guys have string feelings about dependencies, as some folks do, let me know.

Please advise.

Ghostkeeper, 2016-08-02:
Here's the thing. The mesh parsers in CuraEngine are really only used when slicing via the command line. Cura doesn't use it, though some other applications might (notably, RepetierHost). To introduce a new dependency on CuraEngine would have very little gain but relatively large cost.

What would be awesome is if we could get this in the front-end. That would allow Cura to actually load X3D files, so it would introduce a new feature for our own users.
That has a major obstacle though: Cura's front-end is in Python. You'd either have to make a C extension for Python and make sure it compiles with our 4 target platforms (Windows x64, Windows x86, OSX and Linux), or you'd have to translate the entire application to Python, which is possibly very time consuming. We are often willing to take on new dependencies if it provides new functionality, especially if the dependencies are optional and can be left out at the cost of a minor feature.

sevaa, 2016-08-02:
Funny thing, I have an X3D implementation in Python, too :) I wrote it for Blender some time ago. Some data structure translation would be in order, and lots of non-printable stuff will go away. I'm pretty sure I could rejigger that for Cura.

Where's the repo for that, please?

Ocarthon, 2016-08-02:
https://github.com/Ultimaker/Cura

Ghostkeeper, 2016-08-02:
Whoa, awesome! The repository for that is here: https://github.com/Ultimaker/Uranium

The idea is to write a plug-in similar to our other mesh readers. See the 3MF reader as an example: https://github.com/Ultimaker/Cura/tree/master/plugins/3MFReader. You need the getMetaData function in the __init__.py file and a class that extends MeshReader with the read() function.

sevaa, 2016-08-02:
Cloned Uranium.

What's the preferred procedure - just make my changes then push the master back? Private branch, then push that? Fork and work in my own repo?

Ghostkeeper, 2016-08-03:
Our preferred method of including community-provided features is a pull request.

sevaa, 2016-08-03:
Will do.

The build instructions for Arcus and Protobuf could benefit from some guidance for Microsoft Visual Studio users out there. Both build (sort of) fine under MSVC.

sevaa, 2016-08-03:
Got Cura to run by running cura_app.py, but with a funny effect: half the menu items have blank text, and clicking on the "Open file" button does nothing. Looks like Cura has trouble finding some resources (but not others). Windows 7 x64, Python 3.5 x86. Arcus, Protobuf, numpy, scipy, PyQt5 all in place.

Any advice, please?

See the screenshot:

[image]

BagelOrb, 2016-08-03:
@awhiemstra any ideas?

nickthetait, 2016-08-03:
As a workaround you can try drag/drop models from a file manager into Cura.

I actually have the exact same problem on Debian, thought I was the only one and had configured something incorrectly :open_mouth: Have had little time to investigate the cause yet. I do not see an obvious indication of this type of problem in the log, but there are lots of repeated error messages along the lines of:
> QQmlComponent: Component is not ready
> QObject::connect: Parentheses expected, signal MainWindow::
> ... user-interface/resources/qml/Cura.qml:752:5: QML Connections: Cannot assign to non-existent property "onTriggered"
> qml: TableViewSelection: index out of range
> ... user-interface/resources/qml/Cura.qml:75:29: Unable to assign [undefined] to QQuickAction*
> ... user-interface/resources/qml/Settings/SettingExtruder.qml:67:28: Unable to assign [undefined] to QColor

sevaa, 2016-08-03:
I don't see messages like that in the log, but drag&drop works. That should be sufficient for my purposes; I'm not slicing or printing the models anyway.

On a side note. In CuraEngine, the built-in STL reader multiplies every coordinate value by 1000. In UM, it doesn't (as far as I can see). When you import an X3D with shape sizes on the order of 1, the model appears rather tiny on the printer bed. Looks like Cura assumes that the unit of world coordinates is one millimeter - is that so?

The X3D standard explicitly states that the default unit is 1 meter, with an option to override. I guess multiplication by 1000 would be in order, too...

nickthetait, 2016-08-03:
Yes, units are assumed to be mm

Ghostkeeper, 2016-08-03:
They are assumed to be millimetres, but the mesh reading job scales meshes down if they are more than 100 times as large as the build volume. This happens here: https://github.com/Ultimaker/Uranium/blob/master/UM/Mesh/ReadMeshJob.py#L81

awhiemstra, 2016-08-03:
I have seen the same empty menu items a few times. I cannot precisely remember what causes it but I believe it is related to the uranium QML extensions not properly being found. I would recommend looking at those. (they are in UM/Qt/qml/)

sevaa, 2016-08-08:
Submitted a pull request: https://github.com/Ultimaker/Cura/pull/914

I'll add some testing materials to the pull request...

BagelOrb, 2016-10-10:
Merged. Closed.

closed 2016-10-10 by BagelOrb