Ultimaker/CuraEngine #377: MSVC build broken

sevaa, 2016-08-04:
At some point, CuraEngine was compatible with Microsoft Visual Studio. Now it's not. Here are some of the reasons:
- reliance on unistd.h
- "min"/"max" as variable names (they're #defines in MSVC, unless overridden)
- variable length arrays
- reliance on __WIN32 #define - it's not defined anymore, use WIN32 instead

awhiemstra, 2016-08-04:
This is mostly due to MSVC's lacklustre support of C++11. Now that MSVC 2015 finally supports everything we need we can start looking at supporting MSVC again.

sevaa, 2016-08-04:
I'm building with MSVC 2015 Update 3; VLAs are still not there, at least not enabled by default. IIRC, it's a C99 feature, not a C++11 one. Anyways, those can be trivially replaced with alloca().

BagelOrb, 2016-08-04:
I don't think this is currently a problem for us, so I don't we will pick
it up any time soon.

We're always willing accept pull requests though.

sevaa, 2016-08-23:
Addressed, see pull request #383. At least now it builds, I haven't tested properly, since I'm not sure how.

BagelOrb, 2016-10-10:
I'm sorry, we lost sight of this PR.
We will take a look soon; I've added an issue to our internal issue tracker.

jellespijker, 2022-04-12:
Fixed #1639 
Tested on Visual Studio 2019 and 2022

closed 2022-04-16 by jellespijker