Ultimaker/CuraEngine #378: 2.1.2: Gap Filling Method

CCS86, 2016-08-04:
Hi guys, 

I wanted to see if gap filling has been changed in its method, for the next release. IIRC, it leaves the extruder running, and varies head speed to control extrusion rate. Yes? It seems to create a lot of unnecessarily jerky motion:

https://www.youtube.com/watch?v=Z_2ljgJtT8I

The moves in the video are 150mm/s print moves, on a commanded 70mm/s print speed. I think that depending on machine acceleration settings/characteristics, there is a lot of variability in the resulting extrusion.

It really seems like just doing the gap filling moves at the commanded print speed (or better yet, separately configurable), and varying the extruder instead, would give a much better results.

Thanks!

Ghostkeeper, 2016-08-04:
Problem with varying the extruder speed is that it has a gigantic delay due to the tension in the Bowden tube and the pressure inside the nozzle.

BagelOrb, 2016-08-04:
Mark Walker is coincidentally currently on a PR to fix your concerns: https://github.com/Ultimaker/CuraEngine/pull/376

We're going to merge that soon.

CCS86, 2016-08-05:
@Ghostkeeper 

I just don't feel like there is a giant delay. We all know there is some lag in the extruder relative to the print head. But the extruder is stopped any started all around the print. If there was a giant delay, print quality would be very bad. 

To me this means that the delay would be acceptable for gap filling, if it is acceptable for skin.

CCS86, 2016-08-05:
Will that merge make it into 2.2? How far from that release are we, roughly?

BagelOrb, 2016-08-07:
Yes that wil make it into 2.2.

Roughly a month, maximally two.

closed 2016-10-10 by BagelOrb