Ultimaker/CuraEngine #380: CuraEngine's wire protocol calls for vertex duplication

sevaa, 2016-08-11:
In real life meshes, vertices are often used in more than one face. Specifically in 3D printing context, it's even more likely, since meshes tend to represent solid objects as opposed to a collection of unconnected triangles. To treat those meshes efficiently, they usually use a combination a vertex array and a face array, where the latter stores triples of indices into the former.

Notably, Cura front-end's mesh data object has exactly this kind of structure.

The wire protocol of CuraEngine only contains one array - vertices, with no provision for face indices. That's a waste of memory, bandwidth, and CPU.

BagelOrb, 2016-08-31:
Are you talking about wire printing?

Generally you use a very high layer height for wire printing, so I don't think we ever want to optimize for that.

sedwards2009, 2016-08-31:
@BagelOrb He is talking about the communication between Cura front-end and Cura engine and how we represent meshes.

Transmitting the mesh data to the engine is a fairly small amount of time in the whole slicing process. I also doubt that the space/time savings with going to vertex+face lists is really going to add up to much.

BagelOrb, 2016-08-31:
It would be a fairly easy change on the engine side, though.
Wouldn't the frontend change also be fairly easy?

Note that when vertices are read in CuraEngine, vertices which are close to each other (<0.03 mm) are combined into a single vertex in order to save memory and CPU. This happens in `Mesh::addFace(.)`

BagelOrb, 2016-08-31:
PS: an 8MB model already takes 3 seconds to load, but that also includes some other other operations including the vertex melding mentioned above.

Ghostkeeper, 2016-09-04:
Would we want to make the indices in the protocol optional? Not all file formats come with indices. It could be expensive to remove vertex duplication in Python. And if we can't remove duplication, then sending all those indices essentially adds 33% to the amount of data sent to the engine (so 33% longer in "Preparing to slice").

nallath, 2016-09-05:
The removal of vertex duplication is going to be **very** expensive in python. I'm sure that it will be way more expensive than just sending it to the engine and let that handle it. 

We could make the indices an optional parameter though. It would make the sending of OBJ and .3mf files a bit faster.

sedwards2009, 2016-09-05:
I would like to see someone actually measure how long the different phases of the slicing process take before we continue speculating about possible solutions to what might not actually be a real problem.

daid, 2016-09-05:
There is no enhancement to gain here. The engine already finds vertex duplication:
https://github.com/Ultimaker/CuraEngine/blob/master/src/mesh.cpp#L24

The speed in which the model is send to the engine is nowhere near significant in the whole process.

closed 2016-10-10 by BagelOrb