Ultimaker/CuraEngine #384: Infinite slicing loop

Dim3nsioneer, 2016-08-27:
When I put '2mm' into 'minimum travel distance' for retracts, Curaengine of Cura 2.1.3 goes into an infinite loop of 'preparing for slicing' and 'slicing'. It restarts after having sliced about two thirds. It's properly slicing when I put '1mm' in for that parameter.
Maybe it's my installation or a bad combination with other parameters, so if someone could cross-check, that would be great.

Ghostkeeper, 2016-08-27:
Doesn't happen for me. Normal quality on UM2 extended, with just Retraction Minimum Travel on 2mm.

In general, we've found that the infinite slicing loop is usually caused by a race condition in the front-end, rather than a crash in the engine. If the engine crashes, the front-end tends to just keep waiting for progress bar updates indefinitely. We've (tried to) solve this issue for 2.2, though this is one that keeps propping up from time to time.

BagelOrb, 2016-08-31:
Please attach the model with which you have this problem. I suspect it
doesn't happen with any model.

It would also be nice to be able to fully reproduce your settings - can you
export them?

closed 2016-10-10 by BagelOrb