Ultimaker/CuraEngine #387: cmake not working

JustAnother1, 2016-09-04:
In the  README.md  It says (after installing Protobuf and libArc):
`$ mkdir build && cd build
$ cmake ..
$ make`

The first problem is that the generated Makefile is not put into the build folder but one level up.
And then the build fails with this:

lib_CuraEngine.a(settings.cpp.o): In Funktion `cura::SettingsBaseVirtual::getSettingAsFlowTempGraph(std::string) const':
settings.cpp:(.text+0x1506): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::regex_token_iterator()'
settings.cpp:(.text+0x1575): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::regex_token_iterator<3ul>(__gnu_cxx::__normal_iterator<char*, std::string>, __gnu_cxx::__normal_iterator<char*, std::string>, std::basic_regex<char, std::regex_traits<char> > const&, int const (&) [3ul], std::bitset<11ul>)'
settings.cpp:(.text+0x1598): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::operator++(int)'
settings.cpp:(.text+0x15c0): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::operator==(std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> > const&)'
settings.cpp:(.text+0x15e7): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::operator++(int)'
settings.cpp:(.text+0x15f6): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::operator*()'
settings.cpp:(.text+0x1638): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::operator++(int)'
settings.cpp:(.text+0x1647): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::operator*()'
settings.cpp:(.text+0x1716): Nicht definierter Verweis auf `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> >::operator!=(std::regex_token_iterator<__gnu_cxx::__normal_iterator<char*, std::string>, char, std::regex_traits<char> > const&)'
collect2: error: ld returned 1 exit status

Sorry for the German in there: "Nicht definierter Verweis auf " is "Not defined reference to"

Ghostkeeper, 2016-09-05:
Undefined reference to the ++ operator of iterators... Euh...

CMake's "generate" procedure should always put the Makefile in the current directory where it is called from. In this case the CuraEngine/build directory (since you just cd'd into that). Can you make sure you're really in the /build directory when you call `cmake ..`?

JustAnother1, 2016-09-05:
> CMake's "generate" procedure should always put the Makefile in the current directory where it is called from.

It is the default way that most cmke configurations work. But you can do it differently. It is not a thing that happens automatically. cmake makes it easy to have it this way. But it can be configured differently. I assume that is what happened somewhere down in the CMakeList.txt files.

[image]

awhiemstra, 2016-09-06:
Undefined reference to std::regex would suggest you are using a compiler that lacks support for std::regex or that something went really, really wrong with the makefile generation. What platform are you trying to build on?

Ghostkeeper, 2016-09-06:
It's not an undefined reference to `std::regex`, but an undefined reference to an operator on the regex, for some reason.

JustAnother1, 2016-09-06:
I build on Linux Mint. gcc version 4.8.4.
These files should have all the details:
[CMakeFiles.zip](https://github.com/Ultimaker/CuraEngine/files/456612/CMakeFiles.zip)

This is the generated Makefile:
[Makefile.zip](https://github.com/Ultimaker/CuraEngine/files/456606/Makefile.zip)

Shouldn't there be a check done by cmake to test if the compiler has the required features (if that is the issue) ?

Ghostkeeper, 2016-09-07:
Yeah, but we never really bothered to set the `target_compile_features` in our CMake. It could be added at some point.

JustAnother1, 2016-09-09:
It works on Linux Mint 18. gcc 5.4.0.

closed 2016-09-09 by JustAnother1