Ultimaker/CuraEngine #389: Jerk control does not currently work with Repetier firmware

ceryen, 2016-09-08:
Cura currently uses the M205 Gcode to control the value of jerk.  Unfortunately, while this works with Sprinter/Marlin, this doesn't work with the Repetier.  According to Gcode documentation [here](http://reprap.org/wiki/G-code#M205:_Advanced_settings), the M205 Gcode for Repetier will "Output EEPROM settings" rather than set the value of jerk.

For Repetier, it seems that there are two ways to set the value of jerk:

1) According to the documentation above, the M206 Gcode can be used on Repetier to set EEPROM values, which can in turn be used to set the value of jerk in EEPROM.  Their example of "M206 T3 P39 X19.9" sets EEPROM position 39 to a value of 19.9, which supposedly sets the value of jerk to 19.9.  The Repetier source [here](https://github.com/repetier/Repetier-Firmware/blob/90a063ab0d8fb34efb2b5de7ca03e4ff56bfa277/src/ArduinoAVR/Repetier/Eeprom.h) confirms that EEPROM position 39 corresponds to the value of jerk.  Note that setting the value in EEPROM will probably make it permanent rather than temporary.

2) According to Repetier sources [here](https://github.com/repetier/Repetier-Firmware/blob/8987e9127ef7d7dee3dab5c97545c62dc7e5b859/src/ArduinoAVR/Repetier/Repetier.ino#L113) and [here](https://github.com/repetier/Repetier-Firmware/blob/8987e9127ef7d7dee3dab5c97545c62dc7e5b859/src/ArduinoAVR/Repetier/Commands.cpp#L1918), the M207 Gcode can be used to set the value of jerk temporarily rather than in EEPROM.  The M207 Gcode was introduced via [this](https://github.com/repetier/Repetier-Firmware/commit/7f168120e7d690f786b0dcf084da37e13d67f262) merge in Repetier version 0.7.  This was in June 2012 and the Repetier project was started in September 2011, so it's probably safe to assume that most deployed versions of Repetier support the G207 Gcode.  Older versions of Repetier that don't support the G207 Gcode should just ignore it, so it should be safe to use the G207 Gcode with older versions of Repetier, it should just degrade to be a no-op.

ceryen, 2016-09-08:
Note that Repetier is a rewrite of Sprinter and is typically backward-compatible with Sprinter.  It seems that this is one area where Repetier is not backward-compatible with Sprinter.

Since Repetier is typically backward-compatable with Sprinter, existing Repetier users use the "RepRap (Sprinter/Marlin) GCode Flavor in Cura.  Since this case highlights an area where Repetier is not backward-compatible with Sprinter, I wonder if a new GCode Flavor would need to be added to distinguish between Repetier and Sprinter/Marlin.  If a new GCode Flavor is added, would existing Repetier users need to be notified to switch to the new GCode Flavor?

Ghostkeeper, 2016-09-08:
If we would introduce a new g-code flavour for Repetier, you would need to communicate to CuraEngine that this new g-code flavour needs to be used (and thus that it should output M207 instead of M205). That should be fairly simple though. Repetier-Host can just add one extra parameter to their call of CuraEngine. Cura's front-end can add one more setting override to its machine definition.

Anyone calling CuraEngine manually from a command line would have to add `-s machine_gcode_flavor="repetier"` or something to their call.

M207 seems like the best option.

Ghostkeeper, 2016-09-08:
Good research, by the way.

BagelOrb, 2016-09-12:
The engine side is fixed, but I don't know whether the "Repetier" firmare is selectable via the frontend. You could edit the json though...

Ghostkeeper, 2016-09-19:
Repetier flavour was added to Cura:
* https://github.com/Ultimaker/Cura/commit/538a35f81c7ff345448998e34444922d4688ac39
* https://github.com/Ultimaker/CuraEngine/commit/7396deb65bcc3d2b8d2961e51241c27ff30b037b

closed 2016-09-19 by Ghostkeeper