Ultimaker/CuraEngine #391: Inconsistent slicing

brainscan, 2016-09-12:
Cura seems to slice identical objects differently depending where they are on the build plate. This makes fine tuning almost impossible as the results are not consistent. If I put four identical cubes on the build plate and slice, each one will be printed differently. Also the solid infill seems to switch between lines and zig-zag at random so the surface looks different on all of them too. 

Also why does cura insist on adding a +20mm z move at the beginning of every print even if it's removed from the start code. I'm getting fed up having to manually check my code to make sure weird moves don't pop up.

BagelOrb, 2016-09-12:
We need more information; could you please provide:
- your printer model
- the generated gcode
- the Cura version used

As for the inconsistent placing of infill lines: the lines are placed at specific places on the buildplate, but not relative to the object itself. If you want to compare objects, I'd suggest to print them one after another. Also you can supply specific locations for your objects, so that they are exact multiples of the line distance away from each other.
I don't think I've got enough time to fix this issue in the near future.

As for the other problems you mention: please provide the information and I can fix the problems.

brainscan, 2016-09-12:
As for the inconsistent placing of infill lines: the lines are placed at specific places on the buildplate, but not relative to the object itself.

If this is the case then I'll probably just go back to slic3r as I cannot understand how that can possibly work without leaving gaps in the top/bottom surface. So you're basically saying you could waste an entire print because it actually needed to be 0.1 mm to the left or right so the infill meets the perimeter. 

It's a self built reprap printer cura 2.3.0,  I'll upload the gcode when I'm home so you can see the strange behaviour but if the infill is by design then I'll maybe check back later when it's out of beta.

brainscan, 2016-09-12:
Sorry I can't even get cura to open now, it's being really strange. Think I'll uninstall and close this.

closed 2016-09-12 by brainscan

CCS86, 2016-09-13:
The infill will meet / overlap the perimeter regardless of build plate
position.

He was saying that the infill pattern itself is locked to the build plate.

BagelOrb, 2016-09-13:
I think this is what people call 'rage quit'.

You should wait for the actual release. Beta releases are always buggy.

brainscan, 2016-09-13:
I think this is what people call 'rage quit'

Hmmm no I think it's called this doesn't work and I don't have time, no rage involved.

The infill will meet / overlap the perimeter regardless of build plate
position.

So why doesn't it then? And i was talking about solid infill, not infill as I was printing with 0%.  Gaps in the top/bottom layer  and switches from lines to zig-zag infill at random.

BagelOrb, 2016-09-13:
Okay I understand.

It's difficult to understand exactly what you're talking about without
seeing a picture and/or the gcode.

I really need more info before I can answer your questions.

Here's what i _can_ tell:
Filling up thin sections is done by first generating infill lines as normal
and then transforming parallel segments into a single line along the thin
part. You might be experiencing this algorithm kicking in half of the time.
I'm planning to implement a different algorithm which tackles the problem
of thin sections head on without the intermediate representation of lines.
Hopefully that will be in the final 2.3 and hopefully that will solve your
problem.