Ultimaker/CuraEngine #392: run some curaEngine at same time

jiaxiaodong, 2016-09-20:
When I run three CuraEngine at same time, each of them slicing a large stl file about 30MB, each task takes up one hundred percent of cpu resources. Finally there will be a task aborted and given segmentation fault error.
I wonder is this problem caused by my server or the CuraEngine itself. And how to solve it.
Thanks!

here is my profile
system: ubuntu 14.04 LTS (64-bit)
Memory: 8GB
Processor: AMD Phenom II X6 1100T Processor *6

Ghostkeeper, 2016-09-20:
What aborts the task? The segmentation fault?

jiaxiaodong, 2016-09-20:
I'm not sure what aborts the task. Maybe the proessor load is too heavy
There is a "segmentation fault" on terminal when the task aborted. And there will be a empty gcode file.

Ghostkeeper, 2016-09-21:
It's possible you're running into memory problems, too. I can't reproduce it, in any case.

Does the engine write a log, maybe? There's not a whole lot we can deduce without one.

jiaxiaodong, 2016-09-21:
Thanks for your reply. Here is the step to reproduce it. And there are my screen shot of each step
1. open 4 terminal, one of them enter the top commend watch the use of resource, and the others enter the "CuraEngine -v -o a.gcode aa.stl" 
[image]
this picture shows that  each task takes up one hundred percent of cpu resources.

2.After few seconds, the task lower left corner of the screen is aborted, and said and given segmentation fault (core dumped)
[image]

Ghostkeeper, 2016-09-22:
Well, no memory problem either, in any case. Seems to be a bug indeed, though I can't reproduce it yet.

jiaxiaodong, 2016-09-22:
My version is 14.12. 
How about your resourse condition? You may open more than three slicing tasks to reproduce it.

Ghostkeeper, 2016-09-22:
> "CuraEngine -v -o a.gcode aa.stl"

This is going to need at least the "slice" command, and a .def.json file. I used a mesh that was a half-sphere to try and reproduce it, but got no segfault.

Ghostkeeper, 2016-09-22:
> My version is 14.12.
Ah, well there's your problem... That version is two years old. We find and remove a segfault issue in the engine every month or two. I suspect that whatever you're finding has been solved by now.

Perhaps you should try a newer version from software.ultimaker.com. Each Cura version comes with a CuraEngine that can still be run from the command line.

jiaxiaodong, 2016-09-23:
Thank you very much.
I will try new version and show the result in the issue.

closed 2016-10-10 by BagelOrb