Ultimaker/CuraEngine #393: 2.3: Still struggling to fill gaps

CCS86, 2016-09-20:
Hi guys,

I'm just starting to dig into 2.3, and there is a ton of new stuff to check out!

I started with something I have struggled to get satisfying results from in the past: as outer and inner perimeters converge and diverge, Cura Engine has struggled to fill appropriately; ranging from double extruding the gaps, not touching them at all, or adding small dots. I think I was told that the logic was rewritten for this version.

In this print, I have the bottom/top pattern set to concentric, and bottom layers 9999. I am seeing very similar issues. Here are some adjacent layers:

[image]

[image]

[image]

CCS86, 2016-09-20:
In #1, there appears to easily be room for a single .4mm bead. Options to fill would be an over-width single bead, or two passes at a reduced width (which I thought was the new strategy). Instead, the engine only lays in two small squirts.

In #2, it appears to be trying to fill that space, with a single, full-width pass,with little micro extrusions that lay on top of that pass, not centered in the gap.

In #3, there appears to be enough room now to fit 2 full-width passes, and it looks perfect. I just wish the engine could achieve this result consistently.

Ghostkeeper, 2016-09-21:
Only #2 is unexpected for me. In #1 there is most likely not enough room for one pass, except in the two little line pieces due to rounding errors.

Those little pieces in #2 are weird though. Perhaps it calculates that the extrusion should be exactly zero and then still makes the move?

BagelOrb, 2016-09-21:
#2 is exactly the new strategy you talk about in #1
However, the two lines are so close together that the second one reduces to
zero at some places.

The offset of those small lines from the actual gap is smaller than the
difference between the nozzle size and the actual line width, so the
material will still reach the gap.

#1 is something I want to fix soon, but unfortunately it will be after the
final 2.3 release, because it's so difficult to fix.

CCS86, 2016-09-21:
@BagelOrb 

I agree that #2 would probably end up pretty good in printed form. It would just seem that attempting to center that gap filling pass, would yield a better result.

In #1, why does it not attempt the gap filling strategy?

BagelOrb, 2016-09-21:
Good question. The gap filling strategy fails if the infill lines are too
far apart - in its current implementation.

However, it seems like several infill lines aren't even generated! Are you
sure this is from a 100% skin print?

What's the name of this model? I think I already have it on my machine
somewhere..

CCS86, 2016-09-21:
I think it is called "phone side"

Definitely forced to all skin/bottom layers

CCS86, 2016-09-21:
Maybe even tilting the rounding away from the standard, and towards assuming another line will fit, would help #1.

This model is perfectly concentric, and output with a very fine facet tolerance. More lenient rounding could allow that void to be filled. You might end up with a small fraction of over-extrusion, as opposed to major under-extrusion.

CCS86, 2016-11-07:
A thought just popped into my head. I'm not sure if it could be an approach to help:

If a new setting was introduced, "nozzle min diameter", maybe as a percentage of the actual nozzle diameter setting, it could be used as a threshold for evaluating these gaps. 

Say you have a 0.4mm nozzle, and the infill has been generated as above, leaving some gaps that could not be filled at 0.4mm. Is it possible to rescan the infill area at a now reduced nozzle diameter, and treat it just like it was driving a smaller nozzle, only working on the area which was designated infill, but left empty? If you set a 75% nozzle minimum, it would now treat the nozzle as 0.3mm, and attempt to fill these areas.

Any potential here?

nickthetait, 2016-11-07:
While I am naive to the inner workings of CuraEngine, your idea does seem feasible.

BagelOrb, 2016-11-08:
Your algorithm is simple and it would work, but it is computationally wire
expensive.

I've already thought of a way to implement it, the only problem is that I
need time to implement it while I'm to busy with other stuff.

CCS86, 2016-11-09:
That awesome to hear!

Maybe, there could be a way to optimize. If you could take the entire part cross section, then subtract the area filled on the first pass, you would be left with only the gapped areas. The engine could then run only on that area map, with the reduced nozzle diameter. 

Rather than having it work recursively, you could just specify a single reduced nozzle pass.

If that still seems computationally expensive, perhaps leaving it disabled by default. Then once you get all your other settings sorted out, you could flip it on if desired, for the final slicing.

CCS86, 2017-01-09:
Hey guys, I just wanted to let you know that the gap filling in 2.4 is looking great so far, thanks!