Ultimaker/CuraEngine #395: feature request: oversize skin area

sjoerdtimmer, 2016-09-29:
Because some plastics tend to curl up when skin is printed on top of (sparse) infill it would be good if skin were to protrude a little further inside than strictly necessary to support the walls of higher layers.

This is my suggestion to achieve something like that:
```
diff --git a/src/skin.cpp b/src/skin.cpp
index 8846b2d..3c75f1c 100644
--- a/src/skin.cpp
+++ b/src/skin.cpp
@@ -94,6 +94,11 @@ void generateSkinAreas(int layer_nr, SliceMeshStorage& mesh, const int innermost
             }
         }
         
+        // increase size of upskin to better support walls
+        // we grow it at first, and then cut it back to within the bounds of the innermost offset
+        // 4 is a heuristic, should be made configureable
+        upskin = upskin.offset(4*innermost_wall_line_width).intersection(part.insets.back().offset(-innermost_wall_line_width / 2));
+
         Polygons skin = upskin.unionPolygons(downskin);
         
         skin.removeSmallAreas(MIN_AREA_SIZE);
```

This enlarges the skin area by 4 line widths. This should of course be made configurable.
It would be even better to stretch skin to the next infill line. However, that would require heavy integration between infill and skin generation while the above is a simple onelines.

The diff works for me but it is not extensively tested. I don't foresee any problems but I did not try in combination with all possible options or all kinds of weird input stl's.

BagelOrb, 2016-09-29:
I can think of the following problems:
1. a nearly vertical wall gets a 4 line width thick piece of 'top skin'
2. it's computationally quite expensive to do another 2 offsets and an intersection on each part

If you can get the correct result by reordering existing code and/or changing an offset which is already being applied I would be very happy.
You could cache the `part.insets.back().offset(-innermost_wall_line_width / 2)` which is already computed above....

I call such hacking with offsets and intersections "offset magic".
The whole support system thrives on offset magic!

BagelOrb, 2017-07-02:
Skin Expansion is implemented

closed 2017-07-02 by BagelOrb