Ultimaker/CuraEngine #397: 2.3: Back Seam Alignment issue

CCS86, 2016-10-08:
Hi guys, 

I'm trying to make a calibration print to check for gantry squareness. I want to leave the corners all very pure, and don't want seams there. With just a square shape, back seam alignment kept bouncing between back corners. To avoid that, I created a small bulge on the back wall to give the engine a more definitive "back". Now the seam on the very outer perimeter snaps to the back, but the other perimeter seams to not follow. Any way around this?

[image]

Ghostkeeper, 2016-10-10:
I think the seam alignment only works for the outer wall, so far. So we should perhaps formulate this as a feature request?

In our experience, coasting doesn't really work well to reduce the seam yet... Needs some more tweaking, at least for Ultimaker printers.

BagelOrb, 2016-10-10:
To me this seems like a bug.

I'm afraid there is no workaround. :S

sco01, 2017-08-16:
This bug annoys me very much and it still exists in 2.7 Beta. The setting for Z seam alignment currently only works for the outer wall of a shell. The result is that the Z seam of the inner wall(s) often shows thru the outer wall when they are not aligned, leading to bumps and degraded surface finish. The inner seam is also visible when printing with (semi)transparent materials. 

I'm not a developer so I can't contribute much  but it seems like EZSeamType in pathOrderOptimizer.cpp is only honoured for the outer wall for some reason. It may also be related to this bug: [578](https://github.com/Ultimaker/CuraEngine/issues/578)

smartavionics, 2017-08-16:
Hi, to answer your question, I don't think the issue mentioned in #578 is the major culprit here. However, I think that 2.8 could well improve the situation as it should have (currently being evaluated) a brand new wall order optimisation setting. One side effect of the optimisation is that if the z-seam position is specified (for the outer walls), then the inner wall seams should be aligned to the outer wall seams.

Ghostkeeper, 2017-08-29:
We also recently found a bug with how the inside corners are found in the seam alignment (because it should snap to inside corners). I believe @BagelOrb made a fix for that immediately.

smartavionics, 2017-08-29:
Yes, a rework of the seam-snapping-to-corner-engine is underway.

BagelOrb, 2017-08-29:
I don't think this is solved for the next release. All the seam settings still apply only to the outer wall. The other walls start wherever you left off before printing the inner walls. This may be at the end of infill or at the Start XY or closest to a different object, etc.