Ultimaker/CuraEngine #398: 2.3: Concentric Pattern -> One layer of lines?

CCS86, 2016-10-08:
Check out the attached gcode. My settings were for top/bottom pattern: concentric, 10000 bottom layers. For some reason layer #63 is lines instead.

[UMO_Coffee Sifter Body.gcode.zip](https://github.com/Ultimaker/CuraEngine/files/517965/UMO_Coffee.Sifter.Body.gcode.zip)

[image]

BagelOrb, 2016-10-09:
That's pretty weird!

Is that with normal settings (except top/bottom pattern and bottom layer
count)?

Could you please attach the model?

> Check out the attached gcode. My settings were for top/bottom pattern:
concentric, 10000 bottom layers. For some reason layer #63
<https://github.com/Ultimaker/CuraEngine/issues/63> is lines instead.
<https://github.com/Ultimaker/CuraEngine/files/517965/UMO_Coffee.Sifter.Body.gcode.zip>
<https://github.com/Ultimaker/CuraEngine/files/517965/UMO_Coffee.Sifter.Body.gcode.zip>
<https://github.com/Ultimaker/CuraEngine/issues/398>, or mute the thread
<https://github.com/

BagelOrb, 2016-10-10:
I cannot reproduce the problem. Here's what I got:
[concentric_lines_UMO_Coffee Sifter Body_reproduced.gcode.tar.gz.txt](https://github.com/Ultimaker/CuraEngine/files/519071/concentric_lines_UMO_Coffee.Sifter.Body_reproduced.gcode.tar.gz.txt)

Its quite difficult and dangerous to have the beta builds work in a different way from the release builds, because than the changes are never tested during the beta phase. The settings at the end of the gcode are only the settings that you have changed compared to the normal profile.

closed 2016-10-10 by BagelOrb

awhiemstra, 2016-10-10:
The profile appended to the GCode by Cura 2.3 should contain all changes compared to a specified base profile. It should also include which base profile is used. So if you use the High Quality profile and change the top/bottom pattern the GCode should give you a profile that states in its metadata "quality = High" (or something similar) and in its settings "top/bottom pattern = whatever". We then reconstruct the proper profile from this combination of information on import.

BagelOrb, 2016-10-10:
It's a shame we can't see with which nozzle it was sliced or with which material, though.