Ultimaker/CuraEngine #402: change temperature between the first and second layers

cicciom, 2016-10-19:
it would be possible to implement a system like all other slicer to set different temperatures between the first and the second layer?

BagelOrb, 2016-10-19:
Why would you want a different extrusion temperature on the first layer?

BagelOrb, 2016-10-19:
Would you want a higher or lower temperature?

cicciom, 2016-10-19:
I have a membership platform on the glass bed without lacquer, where to join the PLA needs a temperature around 210 ° (the first layer) while others print layer I would set the type to 190-200 ° and I have to do it or manually, by acting on the generated gcode, or from the display. since all the other slicer you have this option you could also implement it you please?

BagelOrb, 2016-10-19:
Ok. I'd like to do this because it's very easy to implement and it will probably help more people than you alone.

If there's any other slicing options Cura is missing, please file new issue reports. We don't have a full overview of what other slicers are doing...

BagelOrb, 2016-10-19:
BTW: what's a membership platform?!?!

cicciom, 2016-10-19:
this platform is called gekofeet I would not want to advertise commercial products but does not need any of the heating plate and lacquer, and for me it is excellent as a solution there is also a discussion forum in http://forums.reprap.org/read .php? 363.642683, page = 1.

thanks to the availability.

It will also be available in repetier host?

BagelOrb, 2016-10-19:
Repetier Host has their own agenda. I don't know whether they're up to date, but I imagine they are lacking behind quite a bit.

cicciom, 2016-10-19:
It is to be able to use the latest version of your windows without repetier how?

Ghostkeeper, 2016-10-19:
You can also do this with the "Tweak at Z" post-processing script. It's a little hidden though:

Go into Extensions -> Post-processing -> Modify g-code. Then add a Tweak at Z script. Set the trigger to Layer Nr, set the layer number to 2, and modify the temperature to your liking.

cicciom, 2016-10-19:
but it is a separate script which is imported into repetier host?

Ghostkeeper, 2016-10-19:
No, the Tweak at Z script is only for Cura's front-end.

cicciom, 2016-10-19:
ok, but I think that having this in the curaengine source which then can be used with all systems I think might be better as a solution

BagelOrb, 2016-11-04:
Fixed in Cura 2.4

closed 2016-11-04 by BagelOrb

VanessaE, 2018-06-17:
Please re-open this, it is NOT fixed, as not only do different filaments need different temperatures, one needs to fine-tune that first layer temperature properly per-filament if the bed has something like PEI or Printbite on it.

VanessaE, 2018-06-17:
Done.  #796