Ultimaker/CuraEngine #404: [master] Segmentation fault when running test suite

thopiekar, 2016-10-23:
```
   dh_auto_test -O--buildsystem=cmake -O--parallel
	make -j4 test ARGS\+=-j4
make[1]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
Running tests...
/usr/bin/ctest --force-new-ctest-process -j4
Test project /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu
    Start 1: GCodePlannerTest
    Start 2: SparseGridTest
    Start 3: LinearAlg2DTest
    Start 4: PolygonUtilsTest
1/6 Test #2: SparseGridTest ...................   Passed    0.01 sec
2/6 Test #3: LinearAlg2DTest ..................   Passed    0.01 sec
3/6 Test #4: PolygonUtilsTest .................   Passed    0.01 sec
    Start 5: PolygonTest
    Start 6: StringTest
4/6 Test #5: PolygonTest ......................   Passed    0.01 sec
5/6 Test #6: StringTest .......................   Passed    0.01 sec
6/6 Test #1: GCodePlannerTest .................***Exception: SegFault  2.61 sec
.[WARNING] Setting an unregistered setting machine_extruder_count to 1
[WARNING] Couldn't load extruder.def.json file for extruder 0. Index out of bounds.
 Loading first extruder definition instead.

83% tests passed, 1 tests failed out of 6

Total Test time (real) =   2.61 sec

The following tests FAILED:
	  1 - GCodePlannerTest (SEGFAULT)
Errors while running CTest
Makefile:64: recipe for target 'test' failed
make[1]: *** [test] Error 8
make[1]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
dh_auto_test: make -j4 test ARGS+=-j4 returned exit code 2
debian/rules:7: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
```
- https://launchpadlibrarian.net/290474630/buildlog_ubuntu-yakkety-amd64.cura-engine_1%3A2.3.99-master~201610222002~rev1337~pkg52~ubuntu16.10.1_BUILDING.txt.gz

BagelOrb, 2017-07-23:
This test has been removed.

closed 2017-07-23 by BagelOrb

mjeveritt, 2018-03-09:
It is customary to refer to a commit which enacts this change. I cannot seem to find a commit entry with a quick search which does this. I'm currently trying to patch out a stabilisation failure for the Gentoo linux ebuild.