Ultimaker/CuraEngine #405: [warning] Couldn't load extruder.def.json from command line interface

qubick, 2016-10-27:
I was playing with previous version of CuraEngine from command line which still works fine,
now trying to explorer new 2.3 features.

It seems like now it shares machine settings with Cura ui, so I am referring machine definition json file like:
``./build/CuraEngine slice -v -j ../Cura/resources/definitions/fdmextruder.def.json -l input/ring.stl -o output/test.gcode
 
But everytime I specify any of def.json file, the Engine shoots me segmentation fault with:
``[WARNING] Couldn't load extruder.def.json file for extruder 1. Index out of bounds.
 Loading first extruder definition instead.
Segmentation fault: 11

If I guess right, is this seg fault from the above warning with out of bound index?
Can anyone help me to resolve this significant fault?

Ghostkeeper, 2016-10-27:
It looks like you're slicing for a dual extrusion printer. It needs two additional files for that: The settings specific for the extruders, such as the offset between the nozzles.

Slicing for the Ultimaker 3, it helps for me to move `ultimaker3_extruder_left.def.json` and `ultimaker3_extruder_right.def.json` from the `extruders` folder in Cura's resources to the `definitions` folder.

Slicing via the command line with dual extrusion takes quite some trickery and shamanistic rattling with boxes of screws in front of the monitor.

BagelOrb, 2016-10-27:
You shouldn't pass the  `fdmextruder.def.json` but the `fdmprinter.def.json` to the engine. The engine should then automatically find `fdmextruder.def.jdon`  if it's in the same folder, or when you have the the environment vairable for it (see `CuraEngine help`)

It shouldn't have segfaulted, but my guess is that it does that because it cannot find any extruder definition...

qubick, 2016-10-27:
Awesome!! Didn't realized until now this was because of such stupid mistake. Thanks a lot

closed 2016-10-27 by BagelOrb

recarens, 2017-12-11:
i solved it moving the extruders.def.json that im using, in my case ultimaker3_extended_extruder.def.json  to de definitions folder.