Ultimaker/CuraEngine #406: Line 757 of slicer.cpp: setting of (bool keep_none_closed) is not considered

fuchstraumer, 2016-10-31:
Writing my own slicing engine by using various parts of a number of slicers. I'm guessing someone had issues with this parameter at some point and just forced the setting, but in that case either it should be removed or keep_none_closed should always be forced to the desired setting and one of the loops removed. 

https://github.com/Ultimaker/CuraEngine/blob/master/src/slicer.cpp#L757

`if (keep_none_closed)
{
    for (PolygonRef polyline : open_polylines)
    {
        if (polyline.size() > 0)
                openPolylines.add(polyline);
    }
}
// And then regardless of keep_none_closed this is performed anyways
for (PolygonRef polyline : open_polylines)
{
    if (polyline.size() > 0)
        openPolylines.add(polyline);
}`

BagelOrb, 2016-10-31:
The feature is actually duplicated across two settings. There's a setting
in mesh_fixes and a setting for mesh_surface_mode and their functionality
overlaps.

I bet the feature is broken as well, but this issue is currently not high
priority because we think not that many people are using it. It is
somewhere on our to-do list - just not that high..

fuchstraumer, 2016-11-21:
Sorry for being late to get back to this.

I figured as such. Just wanted to make sure it got on your guys' radar. Is there a subforum or place I could ask more questions about the engine's architecture? This engine has been my exercise in learning C++ (besides my personal repositories), so it'd be nice ot inquire more and gain more insight.

I'll also add any issues I find, or any fixes I see.

BagelOrb, 2016-11-22:
Here's just fine. No need to go into a forum.

It would be best to have separate github issues for your questions, but
that would probably cause a bit too much overhead.

What did you want to know about the engine architecture?

fuchstraumer, 2016-11-23:
PolygonRef - you describe in the header that this is a wrapped pointer, effectively, and thus doesn't behave entirely like a reference. Would this be a good spot to try to use std::reference_wrapper<Path> ? 

That or I wonder if it might be an apt spot to use a smart pointer. Raw pointers always make me a bit wary, but that's based on my coding background being in the two distinct worlds of Python and embedded systems. 

Also, is this engine meant to be compiled with C++98/99 compilers? It seems to be written in that style, without a lot of usage of C++14 features. 

Thanks - I'm learning C++ by building my own backend and I'd be entirely lost and out of luck without CuraEngine's source being available like this. Credit will absolutely be given where credit is due.

Ghostkeeper, 2016-11-23:
The engine is meant to be compiled with a C++11-compliant compiler. We can't use C++14 yet since some of our compilers don't entirely support it yet (both because we have an old version of GCC to compile for Linux and because MSVC doesn't properly support it yet).

I think the current solution of manually implementing wrappers for pointers/references to certain objects is quite stinky, so the `reference_wrapper` solution sounds interesting.

fuchstraumer, 2016-11-24:
I'm playing with it now - I'll let you know how it works, then. My "fork" of Cura is fairly far removed in some respects, so I'll make a fork later this weekend and keep you updated on that reference wrapper solution. I've also made a lot of changes in the style department: things like always using brackets around for/if statements, and making the casing more standardized (CamelCase for classes + public methods/members, pascalCase for private/protected)

Regardless, I'm standing on the shoulders of Cura's work when it comes to my slicer so I'd like to help out if I can - pay the favor back, if possible.

BagelOrb, 2016-11-24:
Have you had a look at our code standards? https://github.com/Ultimaker/Meta/blob/master/code_conventions.md

We use CamelCase for classes, pascalCase for methods/functions and lower_case for members and variables.
Not all the code is up to standard yet, though. For example the brackets are missing in some cases.

fuchstraumer, 2016-11-28:
*cough* heh *cough*

Probably should have checked those ;p

That's very nearly what I used in the files I had changed. I'll keep an eye on that. I have some ideas for a few features I'd like in my slicing engine that could be used in Cura, as well. I'll use a pull-request for those if applicable.

I didn't work much on the engine over the holiday here in the US, but I played with std::reference_wrapper in my personal projects (namely, my repo "PlanetGen"). I think it could work really well - the syntax is a bit weird because you can't just use "ref.(member_func)", and instead have to use "ref.get().(member_func)". I thought this was me using it wrong at first, but turns out that seems to be intended. 

MSVC is now standards-compliant for C++14 btw. The VS17 Release Candidate applies this. I've been using it, and its been quite the lovely update to VS as an IDE and VS as a compiler. I'm biased though - I have a relative and several friends on that team over at Microsoft :)

BagelOrb, 2016-11-29:
I've reverted part of a faulty refactor around that piece of code: 87a3a30bc5601b8019949c723b65e9eaf08a9c30