Ultimaker/CuraEngine #407: [master] Compilation error

thopiekar, 2016-10-31:
```
   dh_auto_build -O--buildsystem=cmake -O--parallel
	make -j4
make[1]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
/usr/bin/cmake -H/<<PKGBUILDDIR>> -B/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/CMakeFiles /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[2]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
make -f CMakeFiles/clipper.dir/build.make CMakeFiles/clipper.dir/depend
make[3]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
cd /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /<<PKGBUILDDIR>> /<<PKGBUILDDIR>> /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/CMakeFiles/clipper.dir/DependInfo.cmake --color=
Scanning dependencies of target clipper
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
make -f CMakeFiles/clipper.dir/build.make CMakeFiles/clipper.dir/build
make[3]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
[  1%] Building CXX object CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -DARCUS -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/libs  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -static-libstdc++   -std=gnu++11 -o CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.o -c /<<PKGBUILDDIR>>/libs/clipper/clipper.cpp
[  2%] Linking CXX static library libclipper.a
/usr/bin/cmake -P CMakeFiles/clipper.dir/cmake_clean_target.cmake
/usr/bin/cmake -E cmake_link_script CMakeFiles/clipper.dir/link.txt --verbose=1
/usr/bin/ar qc libclipper.a  CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.o
/usr/bin/ranlib libclipper.a
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
[  2%] Built target clipper
make -f CMakeFiles/_CuraEngine.dir/build.make CMakeFiles/_CuraEngine.dir/depend
make[3]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
[  3%] Running C++ protocol buffer compiler on Cura.proto
/usr/bin/protoc --cpp_out /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I /<<PKGBUILDDIR>> /<<PKGBUILDDIR>>/Cura.proto
cd /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /<<PKGBUILDDIR>> /<<PKGBUILDDIR>> /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/CMakeFiles/_CuraEngine.dir/DependInfo.cmake --color=
Scanning dependencies of target _CuraEngine
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
make -f CMakeFiles/_CuraEngine.dir/build.make CMakeFiles/_CuraEngine.dir/build
make[3]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
[  6%] Building CXX object CMakeFiles/_CuraEngine.dir/src/ExtruderTrain.cpp.o
[  6%] Building CXX object CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -DARCUS -DVERSION=\"master\" -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/libs  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -static-libstdc++   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o -c /<<PKGBUILDDIR>>/src/bridge.cpp
/usr/bin/x86_64-linux-gnu-g++   -DARCUS -DVERSION=\"master\" -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/libs  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -static-libstdc++   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/ExtruderTrain.cpp.o -c /<<PKGBUILDDIR>>/src/ExtruderTrain.cpp
[  7%] Building CXX object CMakeFiles/_CuraEngine.dir/src/ConicalOverhang.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -DARCUS -DVERSION=\"master\" -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/libs  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -static-libstdc++   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/ConicalOverhang.cpp.o -c /<<PKGBUILDDIR>>/src/ConicalOverhang.cpp
[  8%] Building CXX object CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -DARCUS -DVERSION=\"master\" -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/libs  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -static-libstdc++   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o -c /<<PKGBUILDDIR>>/src/commandSocket.cpp
[ 10%] Building CXX object CMakeFiles/_CuraEngine.dir/src/FffGcodeWriter.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -DARCUS -DVERSION=\"master\" -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/libs  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -static-libstdc++   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/FffGcodeWriter.cpp.o -c /<<PKGBUILDDIR>>/src/FffGcodeWriter.cpp
[ 11%] Building CXX object CMakeFiles/_CuraEngine.dir/src/FffPolygonGenerator.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -DARCUS -DVERSION=\"master\" -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/libs  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -static-libstdc++   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/FffPolygonGenerator.cpp.o -c /<<PKGBUILDDIR>>/src/FffPolygonGenerator.cpp
[ 12%] Building CXX object CMakeFiles/_CuraEngine.dir/src/FffProcessor.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -DARCUS -DVERSION=\"master\" -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/libs  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -static-libstdc++   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/FffProcessor.cpp.o -c /<<PKGBUILDDIR>>/src/FffProcessor.cpp
In file included from /<<PKGBUILDDIR>>/src/FffPolygonGenerator.cpp:7:0:
/<<PKGBUILDDIR>>/src/utils/algorithm.h: In function ‘std::vector<long unsigned int> cura::order(const std::vector<_Tp>&)’:
/<<PKGBUILDDIR>>/src/utils/algorithm.h:30:5: error: ‘iota’ is not a member of ‘std’
     std::iota(order.begin(), order.end(), 0); // fill vector with 1, 2, 3,.. etc
     ^~~
CMakeFiles/_CuraEngine.dir/build.make:193: recipe for target 'CMakeFiles/_CuraEngine.dir/src/FffPolygonGenerator.cpp.o' failed
make[3]: *** [CMakeFiles/_CuraEngine.dir/src/FffPolygonGenerator.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
CMakeFiles/Makefile2:108: recipe for target 'CMakeFiles/_CuraEngine.dir/all' failed
make[2]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make[2]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
Makefile:163: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
dh_auto_build: make -j4 returned exit code 2
debian/rules:7: recipe for target 'build' failed
make: *** [build] Error 2
```

* https://launchpadlibrarian.net/291523291/buildlog_ubuntu-yakkety-amd64.cura-engine_1%3A2.3.99-master~201610312033~rev1339~pkg82~ubuntu16.10.1_BUILDING.txt.gz

Ghostkeeper, 2016-10-31:
The compiler is not C++11-compliant? `std::iota` is documented [here](http://en.cppreference.com/w/cpp/algorithm/iota) as being in the standard for C++11.

thopiekar, 2016-10-31:
Which compiler are you using?
Compared with a build using Ubuntu Xenial, which works, I only see gcc 6 as a difference here, which probably comes with some changes.
The Ubuntu Xenial build uses gcc-5_5.4.0.

Ghostkeeper, 2016-11-01:
I'm using GCC 4.9.2 on Ubuntu 15.04.

thopiekar, 2016-11-01:
Found it! PR is on the way :running_man:

PR #408 (2016-11-01): Adding <numeric> for GCC 6.x
Otherwise you will get something like:

```
In file included from /<<PKGBUILDDIR>>/src/FffPolygonGenerator.cpp:7:0:
/<<PKGBUILDDIR>>/src/utils/algorithm.h: In function ‘std::vector<long unsigned int> cura::order(const std::vector<_Tp>&)’:
/<<PKGBUILDDIR>>/src/utils/algorithm.h:30:5: error: ‘iota’ is not a member of ‘std’
     std::iota(order.begin(), order.end(), 0); // fill vector with 1, 2, 3,.. etc
     ^~~
```

* Fixes #407
closed 2016-11-01 by Ghostkeeper