Ultimaker/CuraEngine #409: path order in a layer

ynilsiam, 2016-11-07:
When I try to print a gear, It starts in the middle of the gear first(red), then the inside(green) and last the outside(black).
Is there anyway to make it goes around from outside to inside or wise versa ? any configure in fdmprinter.json?
Thank you.
[image]

BagelOrb, 2016-11-07:
Yes. There's a setting for Infill Before Walls and a setting for Outer Wall
Before Inner Walls.

They are present in fdmprinter.def.json, but you can also simply edit them
from the cura frontend by making the settings visible first in the
visibility preferences.

ynilsiam, 2016-11-07:
Thank you very much for answering.
However, would you please provide the parameters' names you mentioned?
I looked in fdmprinter.json, but couldn't find them.
Thank you again.

-- 
Yuenyong Nilsiam
PhD Candidate
Michigan Tech Open Sustainability Technology Lab
<http://www.appropedia.org/Category:MOST>
Department of Electrical & Computer Engineering
Michigan Technological University
EERC Building 121
1400 Townsend Drive
Houghton, MI 49931-1295
615-378-7756 <+16153787756>

BagelOrb, 2016-11-07:
I don't know by heart.

Ctrl+F "Before"

Or ask me again tomorrow (I'm in Amsterdam time zone)

Ghostkeeper, 2016-11-08:
The setting ID is `outer_inset_first`. Its label is "Outer Before Inner Walls".

Ghostkeeper, 2016-11-08:
That setting just determines the order in which it prints the walls though.

It doesn't determine the order in which to print the macro-scale parts. There is no setting to determine that, really. It depends on which order leads to the shortest travel moves and the fewest extruder and flow switches.

BagelOrb, 2016-11-08:
The other ID is `infill_before_walls` It's label is "Infill Before Walls"

ynilsiam, 2016-11-08:
Thank you very much. However, I am using version 15.06, there is no that
parameter.
Is there anyway to do that by edit the code?

BagelOrb, 2016-11-08:
Why are you still on Cura 15.06? Cura 15.06 is a beta release.

Please update to Cura 2.3.1 That's the newest version.

ynilsiam, 2016-11-08:
In my understanding, the newer version does not support RepRap printer any
more.

BagelOrb, 2016-11-08:
Cura 2.4 will support all printers which Cura 15.04.6 supported.

Cura 15.06 supports less than 2.3, but you might be confusing 15.06 with
15.04.6

Cura 2.3 sure support custom printers...

ynilsiam, 2016-11-08:
> auth/AIe9Ea3nYT_2pz_

I only use CuraEngine, not the GUI.

BagelOrb, 2016-11-09:
CuraEngine does always support RepRap printers.

You should definitely check out the engine of Cura 2.3.1...

ynilsiam, 2016-11-09:
I see. Thank you very much.

ynilsiam, 2016-12-08:
I checked the CuraEngine 2.3, but I could not find infill_before_walls.
Which class should I look into?

ynilsiam, 2016-12-08:
Is there anyway to fix this in the code?
Thank you very much.

Ghostkeeper, 2016-12-09:
It is used here: https://github.com/Ultimaker/CuraEngine/blob/master/src/FffGcodeWriter.cpp#L840 (and a few lines below that again).

All settings are defined here: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json

ynilsiam, 2016-12-12:
> It is used here: https://github.com/Ultimaker/CuraEngine/blob/master/src/
> FffGcodeWriter.cpp#L840 (and a few lines below that again).
>
> All settings are defined here: https://github.com/Ultimaker/
> Cura/blob/master/resources/definitions/fdmprinter.def.json

Thank you very much.

mokusolo, 2016-12-22:
I have a problem about path.And I'm not sure if it has something to do with the para "infill_before_walls".
I'm using cura 2.3.1 to slice a pillar,then I put it in simplify3D to preview.
It shows that after some layers the start point changed to B but later(not sure about layers)it changed back to A.
[image]
[image]
I confirmed the z seam alignment was "shortest".
So confused.I don't think such a regular form of model can have two different start points.It may affect printing quality.
Anyone can give me a hand,Please?
Thank you for reading.

Ghostkeeper, 2016-12-23:
What often happens is that it appears to alternate two layers in the order in which the infill lines are drawn, and as a result where it starts a wall.

The Z Seam Alignment setting should take care of (only) the outer wall. It is a combination of what the shortest travel move would be and some 'snapping' to a nearby concave corner if one is available. But there is rounding involved here. CuraEngine works with micrometre accuracy, but when it comes to calculating angles and distances there is sometimes a little more rounding involved. This rounding can cause some other corner to be considered optimal.

For a shape such as this, I'd consider setting Z Seam Alignment to "Back" better. Or in 2.4, setting it to the new manual position.

mokusolo, 2016-12-24:
Thank you for your advice :)
I'll try "Back" next week.
And maybe I should work hard on CuraEngine.It's fantastic.

ynilsiam, 2016-12-24:
How can I change to make the beginning of path start from the edge, not the
middle if LayerPath?

Ghostkeeper, 2016-12-27:
Start from the middle of a layer path? What do you mean?

ynilsiam, 2016-12-27:
[image]

You can see that it starts from the middle of the layerpart.

BagelOrb, 2016-12-30:
I'm not sure I understand, but you can set the setting to print he outer
wall before inner walls, so that it will start each part on the outside.

ynilsiam, 2016-12-31:
How can I do that in CuraEngine?

-- 
Yuenyong Nilsiam
PhD Candidate
Michigan Tech Open Sustainability Technology Lab
<http://www.appropedia.org/Category:MOST>
Department of Electrical & Computer Engineering
Michigan Technological University
EERC Building 121
1400 Townsend Drive
Houghton, MI 49931-1295
615-378-7756 <+16153787756>

BagelOrb, 2016-12-31:
There's a setting called something like outer_before_inner_walls or
something. I can give a definitive answer when I get back from my
holidays...

> I'm not sure I understand, but you can set the setting to print he outer
> wall before inner walls, so that it will start each part on the outside.
>
>
> > [image: screenshot from 2016-12-27 16 46 07]
> > <https://cloud.githubusercontent.com/assets/8889952/21509107/fee5b580-
> cc53-11e6-9f04-4e6899a87cb9.png>

> >
> > You can see that it starts from the middle of the layerpart.

> > <https://github.com/Ultimaker/CuraEngine/issues/409#
> issuecomment-269387214>,

> AIe9EcnbXRHyVrPsPHVZsQG5iE8YwTiMks5rMYdhgaJpZM4Krm_s>

> <https://github.com/Ultimaker/CuraEngine/issues/409#issuecomment-269834187
>,
AIemYMMC62swOLXX00py2Z7fgGgUdyRxks5rNZIZgaJpZM4Krm_s>

-- 
Yuenyong Nilsiam
PhD Candidate
Michigan Tech Open Sustainability Technology Lab
<http://www.appropedia.org/Category:MOST>
Department of Electrical & Computer Engineering
Michigan Technological University
EERC Building 121
1400 Townsend Drive
Houghton, MI 49931-1295
615-378-7756 <(615)%20378-7756> <+16153787756 <(615)%20378-7756>>

Ghostkeeper, 2017-01-02:
> The setting ID `is outer_inset_first`. Its label is "Outer Before Inner Walls".

It was mentioned earlier in this thread.

closed 2017-01-06 by BagelOrb