Ultimaker/CuraEngine #410: writeInt2mm produces unprocessable values

erwint, 2016-11-09:
Since 2.3.0 release I noticed my printer would not correctly start printing, as the bed would not come back up after the initial homing sequence. It worked with 2.3.0 beta, so I browsed all changes and suspect it got broken by dce8a4629f811f850344acf5c322023633719761,  which will cause low mm values to be formatted like `Z.2`(instead of `Z0.200`). I verified manually changing all those directly in the .gcode by adding a `0` in front of the decimal point (e.g. `Z0.2`) fixes the problem

BagelOrb, 2016-11-09:
What printer do you have?

Do you know what type of firmware it uses?

erwint, 2016-11-09:
A Qidi X-One, I have no idea what firmware it uses.

Ghostkeeper, 2016-11-10:
It probably has home-cooked firmware. I couldn't find it either. All I found was that a bunch of people tried to put Sailfish on it.

Ghostkeeper, 2016-11-10:
I wouldn't expect CuraEngine to leave out the initial zero before the radix, though.

daid, 2016-11-11:
Caused bugs for the UM3 firmware initially as well (have now been fixed)

BagelOrb, 2016-11-11:
I didn't know this caused problems for Griffin as well.

I've changed it for Cura 2.4
See https://github.com/Ultimaker/CuraEngine/commit/4aa1cc47f4f4f0b1163d27aa1fd1615f6fcf8356

Generally there will now be a stagering 7 more characters in the gcode! :stuck_out_tongue:

closed 2016-11-11 by BagelOrb

daid, 2016-11-11:
Single wrongly wrote regexp to lift numbers from GCode parameters.

BagelOrb, 2016-11-12:
@daid I don't understand what you mean