Ultimaker/CuraEngine #413: Struggling to understand why the slicer outputs [WARNING] with a list of all settings

slavicd, 2016-11-16:
The build is from master (same happens with 2.3 htough).
My output after 
`./CuraEngine slice -v -j "def.json" -o "outfile.gcode" -l "input.stl"`
contains a warning line like this: "[WARNING]  -s support_z_distance="0.15" -s infill_overlap="20" ......" witha  long-long list of settings.

I need to know why this happens because my CuraEngine ouput differs from output of UI Cura.

BagelOrb, 2016-11-16:
That actually shouldn't be a warning, it's just the info of which settings
you have used.

The reason it's a warning is that we want it to be outputted even with
standard logging level. We should fix that...

You pass "def.json" to the engine, but u do know this should be the
definition file of your machine, right? So for me that would be
"ultimaker3.def.json"...

Ghostkeeper, 2016-11-16:
Engine output is actually logged as [DEBUG], not [WARNING]: https://github.com/Ultimaker/Uranium/blob/master/UM/Backend/Backend.py#L89

BagelOrb, 2016-11-17:
Yes, but the engine itself outputs it as [WARNING], the log shows:
"[Backend] [WARNING] -s ...."
So the frontend adds the [Backend] and the backend adds the [WARNING]

slavicd, 2016-11-21:
What does the last output line mean? "Filament: 105055" ? Is that length or volume?

BagelOrb, 2016-11-21:
That's volume in mm^3

BagelOrb, 2016-11-21:
The original issue still stands: CuraEngine outputs the used settings as a WARNING

slavicd, 2016-11-22:
It seems to be related to the same thing: it turns out all the output from CuraEngine command is done to stderr, rather than stdout, which is quite an issue for me to grab it from inside some kind of exec() function in PHP or anything else for that matter.

Try `./CuraEngine help`, for example

BagelOrb, 2016-11-23:
User feedback is done over std::cerr, while gcode output is done over std::cout (if not otherwise specified)

You could pipe `CuraEngine -j ... -l model.stl | output.gcode`

BagelOrb, 2017-07-23:
I think the fact that the settings are being output as WARNING isn't an issue which is important enought to ever solve.

closed 2017-07-23 by BagelOrb