Ultimaker/CuraEngine #414: problem on cooling settings

mokusolo, 2016-11-20:
I have some trouble with cura 2.1.2 when I want to change cooling settings.
I'm preparing to print without any fan to test.In my opinion,when the allow cooling is false,it means no fan speed all over the printing process.But the gcode has no difference with that when I set the value true.So I'm quite confused.Did I use it incorrectly?Or I misunderstood the value meaning?
Can anyone help me please?Thank you.
I searched on fan speed problem in issues.There are few about this.And it doesn't help much.

BagelOrb, 2016-11-20:
First of all, try the latest version: Cura 2.3.1

Hopefully the bug is fixed there...

A workaround for the bug is set the fan speed to zero while keeping it
enabled.
Hope that helps!

Ghostkeeper, 2016-11-21:
If I remember correctly, 2.1 had this bug where the engine only looked at the fan speed and the front-end would set the fan speed to 0 if the fan is disabled.
Problem was that some of the profiles for new materials would specify a different fan speed in the quality profile. This then prevented Cura from setting the fan speed to 0 when the fan was disabled. This was recently fixed here: https://github.com/Ultimaker/CuraEngine/commit/c0495edf48f4df75456b6c1ffb738d0fdb36583e

Getting version 2.3.0 or later should fix the problem.

closed 2016-11-21 by Ghostkeeper