Ultimaker/CuraEngine #416: Feature Request - add info to gcode so that DuetWifi controller may report progress

smartavionics, 2016-12-06:
I have a Kossel delta controlled by a DuetWifi and am getting some nice results using Cura. One irksome thing is that the Duet's progress report feature doesn't work with the gcode produced by Cura. The Duet can predict how long the print will take based on either the amount of filament used so far, the number of bytes of gcode consumed so far and the time taken to print the layers so far.

Is is possible for you to put a comment into the gcode file that reports the total amount of filament used? I haven't used previous incarnations of Cura, just the latest, but I believe that info used to be present in earlier versions?

Thanks,

Mark

Ghostkeeper, 2016-12-07:
I can tell you in advance that we won't be able to add this feature without it being initially implemented in the community. For two reasons:
* If we make a feature request to get this implemented, it would get rejected immediately or postponed indefinitely because it doesn't help Ultimaker with meeting its deadlines.
* We have no Kossel printer at work to test with.

We can however accept pull requests from the community and guide people towards implementing it.

smartavionics, 2016-12-07:
Understood. In that case, I shall have a go at implementing it myself.

Ghostkeeper, 2016-12-07:
The total filament usage so far is stored per-extruder in the class `GCodeExport::ExtruderTrainAttributes::totalFilament`.

BagelOrb, 2016-12-07:
I don't understand your feature request. Cura already does this!

Perhaps the way in which cura writes the filament used is not the format
the kossel firmware is expecting..?

Please show how your printer expects the start of the gcode to look like.

smartavionics, 2016-12-07:
The DuetWifi code is looking for a string that says "ilament used" followed by the length in mm. You can find the code in https://github.com/dc42/RepRapFirmware/blob/dev/src/PrintMonitor.cpp

Maybe the previous versions of Cura output that and the current version doesn't?

BagelOrb, 2016-12-07:
Legacy Cura (1.15.04) outputs the following:
```
;Filament used: 0.816m 6.0g
```

Which is in meters.

Cura 2.3 outputs the following for an Ultimaker 2 family printer:
```
;MATERIAL:3748
;MATERIAL2:0
```

But nothing is outputted for other printers than UM3 or UM2 family printers.

You can add it on `gcodeExport.cpp::getFileHeader(.)`. On line 201 you should add an `else` case.

smartavionics, 2016-12-07:
Thanks a lot for the info, I shall take a look at that.

BagelOrb, 2016-12-07:
Good luck!

smartavionics, 2016-12-07:
OK, so I have the following snippet working in GCodeExport::getFileHeader(), it converts the volume into a length as the Duet is expecting that rather than volume. I will submit a pull request if you think this is along the right lines...

`        else if (flavor == EGCodeFlavor::REPRAP)
        {
            prefix << ";MATERIAL:" << ((filament_used.size() >= 1)? static_cast<int>(filament_used[0] / extruder_attr[0].filament_area) : 6666) << new_line;
            prefix << ";MATERIAL2:" << ((filament_used.size() >= 2)? static_cast<int>(filament_used[1] / extruder_attr[1].filament_area) : 0) << new_line;
        }
`

BagelOrb, 2016-12-07:
Please do submit a PR, we can always discuss further there.

This code doesn't do what you day, though.
It doesn't convert from mm^3 to mm and it doesn't output `Filament used:` but uses the same syntax as the UM2 does.

Didn't you accidentally post an outdated version of your code?

smartavionics, 2017-05-10:
This was implemented, closing...

closed 2017-05-10 by smartavionics