Ultimaker/CuraEngine #418: Extraneous support material added when placement is "Touching Building Plate"

smartavionics, 2016-12-09:
Hi,
I am seeing an issue where a small pad of support material is incorrectly added under a region which would have support added higher up in the print if placement was "Everywhere". If I set the placement to "Everywhere" this pad goes away and the support is added above when the void is reached.

So, here's a screenshot of the pad that shouldn't be there. It's the light blue region on top of the darker blue area.

[image]

And here's a screensot showing what is above that area. Basically, there's a solid region that extends over the whole supported region and then on one side of that is a taller solid region that has a hole in it. It's the area under the hole that gets covered by the unwanted pad.

[image]

BagelOrb, 2016-12-09:
I don't really understand.

Did you color this patch yourself?

Is this patch on a different layer than the big square of support?

Could you please attach your model to this issue?

This might very well have to do with a bug I solved for Cura 2.4 in which small patches of overhang would create support on every layer, since at the top of the round hole there is a really small overhang area. The same doesn't hold for the top of the hexagonal hole, so I'm quite surprised to see this.

smartavionics, 2016-12-09:
No, I didn't colour the patch, that's what Cura is displaying

Yes, the patch is above the big square of support  - it is printed on what should be an empty layer (between support and the model).

I have copied the model to

[https://www.dropbox.com/s/8ddm5n5wrz67nfr/KX_magni_combo_mount.gcode?dl=0](url)

smartavionics, 2016-12-09:
Oops, here's the stl.

[https://www.dropbox.com/s/t15tyvs33f4uxr4/magni_combo_mount.stl?dl=0](url)

Ghostkeeper, 2016-12-09:
It's displaying the top few layers in a brighter colour. That's why that patch shows up brighter.

smartavionics, 2016-12-09:
Sure, but the patch should not be there at all!

That layer should be empty in the support region because it's the gap between the support and the area above that is being supported.

Ghostkeeper, 2016-12-09:
I can't reproduce it exactly (I tried also with Kossel Mini but I don't have the exact same printer).

But it does seem to produce a little patch there:
[image]

smartavionics, 2016-12-09:
Also, there's something odd about the interface region below where the patch occurs, as shown here

[image]

See that weird triangular feature. It has offset the filament strands. You can see that clearer using the gcode viewer in s3d

[image]

BagelOrb, 2016-12-09:
I could reproduce it on 2.3, but not in the Cura 2.4 pre-beta, so this bug has been solved.

The triangular pattern is just created because it wants to print support a little bit further out at that spot. That's a bit ugly; the only reason there is more support there is to eventually support the round hole, but it cannot support it due to the restriction of support only being placed on the build plate. I wouldn't call that a bug.

closed 2016-12-09 by BagelOrb