Ultimaker/CuraEngine #422: Compiling CuraEngine

ghamrawyk, 2016-12-14:
Hello everyone, 
I am new to CMake and to C++ in general and I wanted to compile CuraEngine to generate any Gcode file and get familiar with the whole environment. I am on Windows 7 professional 64-bit, and I am using CMake to compile, build, etc... 

What I did so far is:  - I cloned CuraEngine successfully
- I cloned and installed protobuf successfully. (Tested it using nmake check)
- I downloaded libArcus (but I am not sure if I installed it correctly). 
When I type "cmake .." to compile CuraEngine I get this error:
 
_**CMake Error in CMakeLists.txt:
  Imported target "Arcus" includes non-existent path

    "D://include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.**_

Please, any suggestions on what I should do, check, or maybe reinstall. Is there a way to test whether everything is installed properly or not? I am very confused as I am new to this, so any simple instructions would do. 

Thank you very much in advance,

Ghostkeeper, 2016-12-15:
There's a couple of things you need to ensure.

You need to make sure that Arcus is properly "installed" somewhere. When you compiled Arcus with CMake you had an option in CMake `CMAKE_INSTALL_PREFIX`. This is where it installed Arcus. It installs several things there: The header files (.h) and the library files (either .a or .dll). Check that they are there.

Then you need to make sure that the CMake project of CuraEngine can find those files. There's the Arcus_DIR variable for that (which is a little non-standard; usually you'd have ARCUS_INCLUDE_DIR and ARCUS_LIBRARY or something).

Alternatively you can disable the piece of CuraEngine that communicates via sockets by setting ENABLE_ARCUS to "OFF". It won't need Arcus or Protobuf then, but you won't be able to use CuraEngine via our Cura front-end (or any other Protobuf interface).

ghamrawyk, 2016-12-15:
Thank you very much for your quick answer. When I installed it, i made sure to set the CMAKE_INSTALL_PREFIX to an Arcus folder, which now contains everything. Now I searched there for the header files and the lib files. Apparently there are header files but no lib files. I didn't find any .dll or .a files. On a side note, when I installed protobuf, the lib files were in .lib format (object file Library) "not .dll nor .a". Even though, in Arcus, this doesn't exist. Any suggestions on what I should do? 

I tried to compile it with Arcus set to OFF, and the generation was completed. As far as I understand, I should find the generated file in the build folder in the CuraEngine path, is that right? What is the expected output file format? How can I generate a GCode file?

Thanks again, your help is much appreciated.
Cheers,

ghamrawyk, 2016-12-15:
I would also like to add that I am using the CMake GUI, and I also made sure that the Arcus_DIR is set to the Arcus folder (D:/PhD/Cura/Arcus). But for some reason when I am generating in CuraEngine build, it changes the Arcus_DIR "automatically" to CuraEngine folder (D:/PhD/Cura/CuraEngine), which isn't the one that is supposed to have all the headers and libs for Arcus. 

Thank you,

Ghostkeeper, 2016-12-15:
I'll explain the process with CMake a little more extensively. I hope you don't mind me taking you by the hand a bit. I don't know how much you know.

1. First you typically open CMake-GUI and point it to the directory that contains the CMakeLists.txt file. You seem to have done this successfully. It's good practice to set "where to build the binaries" to some other directory, like a subdirectory called "build". This way you won't contaminate the source tree with compiled files, and it'll be easier to reset everything by just deleting that build folder.
2. Then you click "Configure". This runs the CMakeLists script, executes the code in there. CMake has its own little scripting language. You don't really see anything as a result from this. It only generates a bunch of files which it puts in a CMakeCache folder.
3. Then you click "Generate". The purpose of this is to generate the files that your preferred compiler needs to create the end result. In the beginning you'll have been asked which compiler you prefer: MinGW, Visual Studio, CLang, etc. CuraEngine has been tested with MinGW (preferred compiler) as well as Visual Studio. From the sound of things you've chosen Visual Studio. The result of the Generate command is then a Visual Studio Project File.
4. Then you need to invoke your compiler, separate from CMake. With GCC on Linux that's done with the "make" command. With MinGW it's "mingw32-make". With Visual Studio, your option, it's typically done by opening the generated Visual Studio project and clicking on the compile button in there. The end result of this is an actual binary. In CuraEngine's case that's just a .exe file.

> On a side note, when I installed protobuf, the lib files were in .lib format (object file Library) "not .dll nor .a".

Ah, that's right. ".lib" is another one of those formats. It works just as well. I think it depends on the compiler and whether you choose for static linking (library gets packed into the binary) or dynamic linking (library stays a separate file).

> I tried to compile it with Arcus set to OFF, and the generation was completed. As far as I understand, I should find the generated file in the build folder in the CuraEngine path, is that right? What is the expected output file format?

So to answer your questions directly, the "generating" should create a Visual Studio project file. If you open this file in Visual Studio and click compile, the end result should be a .exe file, which is CuraEngine that you can run. Running CuraEngine via a command line will give you instructions on how to use it; you basically say where it needs to find a 3D model, where it needs to find settings to slice with, and where it needs to put the resulting g-code.

> Even though, in Arcus, this doesn't exist. Any suggestions on what I should do?

So probably, if you didn't go past generating, you haven't actually compiled it yet by opening the Visual Studio project and compiling in there.

> When I tried to type the make and make install commands, I got this: 'make' is not recognized as an internal or external command, operable program or batch file. As far as I understand, these are linux commands.

That's right. Below that bit in the readme though there are also instructions on how to build on Windows. They say to use "mingw32-make" there, which are correct instructions for the MinGW compiler. For you that's via the Visual Studio project file again. Though I don't know how well this has all been tested with Visual Studio's compiler.

If you didn't mean to use Visual Studio, then you can go into CMake-GUI's application menu into File -> Delete Cache. The next time you configure you'll be asked to select your generator again. And you'll want to choose MinGW then. After that the result of generating will be Makefiles, and you can compile in a command line by typing "mingw32-make", instead of by going into some Visual Studio project.

ghamrawyk, 2016-12-15:
I can't thank you enough, you truly helped me. I hope this little tutorial would be a trigger for something good. 
Thank you again, I really appreciate the time and effort you put to write all this. Now it's much clearer. 

Cheer,

closed 2016-12-15 by ghamrawyk