Ultimaker/CuraEngine #423: Feature Request: New Bed Adhesion Technique: Spider Fingers

cinderblock, 2016-12-19:
I'm having issues with edge/corner peeling on some prints. I've bumped up my raft margin plenty and don't want to keep cranking it up and waste a bunch of filament.

In observing these peel-ups, they only seem to happen on relatively convex edges once more than one layer has been added, seemingly to do with new layers shrinking as they're added.

One way I thought would be a good possible fix for this is an extra brim for the raft specifically. This has basically been requested in another issue https://github.com/Ultimaker/Cura/issues/942.

While thinking about this problem more, I had a seemingly new idea for a possibly better bed adhesion technique:

## Spider Fingers

> [image]
> *weeee mspaint!*

The implementation would be a bunch of single layer sticks that stick out normal to the edge of the base layer.

The idea is that since edges and convex corners peel the most we want to reduce those as much as possible. These sticks make the peeling happen ideally only at their distal end where we really don't care and they don't get to peel under the real part.

Once the print is complete, the fingers should be able to break off because they're thin enough.

Possible options would include:
 - Finger length
 - Finger spacing
 - Finger width
 - Fingers wall overlap (intentional interference between fingers and part wall base)
 - Finger teardrop proximal (fanout at end of fingers close to part)
 - Finger teardrop distal (fanout at end of fingers far from part)
 - Finger perimeter (add a perimeter, like a skirt, that the fingers all connect to)
 - Finger layers (default 1)
 - Finger smear (at the distal end of fingers, before retraction, drag the end of the finger away from the part to break filament and prevent the head from accidentally peeling up the just placed finger.)
 - Finger direction (start extruding close to or far from part)

Ghostkeeper, 2016-12-20:
Worth a try as an experiment. It is sort of the opposite of what the brim is going for though.

The major cause for warping at the corners is shrinkage of the material. It also tends to happen more when lines are in parallel, because they are then straining all in the same direction. When using brim, the shrinkage of the brim lines causes the brim lines to pull more tightly together rather than pull on a loose end, so shrinkage has less of an effect then. This is also why concentric skin is so effective at increasing build plate adhesion.

But I don't think this spider technique has ever been actually tried. It might 'absorb' some of the stress pulling from the inside of the model. Who knows?

BagelOrb, 2016-12-20:
I never experienced problems with the raft coming loose.
What printer do you have? What material are you using?

Do you think your technique would work better than just increasing the raft
margin include the area which would have been covered with fingers?

Note that the middle of your picture shows an implementation problem: what
to do with overlapping fingers? Imagine a small deep crevice; the only
fingers possible in that crevice would overlap with each other or with the
model..

cinderblock, 2016-12-20:
I'm running an Airwolf HDx (free to me since it was "broken"; just a failed limit switch) via OctoPrint. Printing ABS. But that, imho, has no relevance on if this idea has any merit. I've been asking around a little and have not heard of anyone even trying this method.

One of the problems I've been having is that the raft itself starts to peel (ever so slightly) even before the part is started. That of course grows once the part does start. This never happens with a brim but I don't trust my bed's level and prefer the finish I get from a rafted part. This to me indicates that it's the thickness of the raft that is causing the initial peeling which is why I want to add a brim to my raft which started this whole thought process.

---

Ultimately however, yes, I think fingers would be better than a brim. If one finger peels, it doesn't spread to the adjacent fingers immediately. This is the opposite of what a brim does. If the brim / raft starts to fail, it will keep peeling and spread to adjacent sections. The hope is just that it doesn't start peeling in the first place.

Specifically, I think this method would be able to achieve the same level of bed adhesion as a brim with drastically less material.

---

I was trying to show a method for dealing with concave corners in my picture. Unfortunately, that intent was not completely captured in mspaint. Clearly, overlapping with the model would be a no-no.

A simple technique for terminating fingers that would otherwise be too long: grow each line until it hits another line or it's max length is reached. (implementation details would probably vary for performance reasons)

---

A couple small thoughts I forgot to mention originally:

 - Option to have all fingers be radial from some central point instead of normal to the edge
 - Option to have all fingers radiate from a virtual line inset from the edge ~1mm. This would be a decent middle ground between all normal or all fingers radiating from a central point.

Specifically, these options are intended to be used to be able to replicate the finger equivalent of "mouse ears" (extra brim at sharp convex corners).

BagelOrb, 2017-01-10:
I do think the idea has some merit, but it is quite a hassle to implement.
Of course you are free to do so and set up a pull request to get your code changes merged.

In the mean while I am putting two new features on our priority list:
- mouse ears
- raft chamfer: inset each raft layer a bit to combat the peeling

nickthetait, 2017-01-10:
I'll second the mouse ears. This is a technique people have been manually designing since the very early days of rep-rap.

Not as convinced on usefulness of raft + brim. Though I have heard this requested by one LulzBot user  :neutral_face: