Ultimaker/CuraEngine #424: confusing paths when choosing Spiralize Outer Contour

mokusolo, 2016-12-26:
Actually I searched key word "Spiralize" in issues.But my problem maybe little different from the #214 so I made a new issue.
I'm trying spiralizing a model with Cura2.3. It's like this.
[image]
After slicing I previewed,it looks like the 3 layers in following screenshot(red color) didn't actually print.Or I have to say with such high speed it wouldn't print well.
[image]
[image]
I'm new,and sorry to that I don't know how to locate the gcode of the specific few layers.To count layer by layer is not scientific...
Then I also sliced it without choosing spiralizing. Maybe the similar paths?
[image]
I doubt if the model has defect.But magics show nothing...

Ghostkeeper, 2016-12-27:
In Repetier-Host you can select pieces of g-code and it'll highlight that g-code in the 3D view. I often select from the beginning and expand the selection using Shift+PageDown to find the piece of g-code that I need to find. Perhaps you could try that to see if the g-code is actually faulty and not the viewer. In the g-code, a travel move is indicated by `G1 X... Y... F...` where the number after the `F` indicates the speed in millimetres per minute. 7200mm/min is usually as high as it will go (120mm/s).

Otherwise you can upload the g-code here (in a .zip) and have one of us find it for you.

Spiralize does some tricky stuff with the layer borders that your viewer might not handle.

BagelOrb, 2016-12-30:
Please attach the gcode and STL file in a zip, so that we can see what the problem actually is.

mokusolo, 2016-12-31:
So sorry that I'm on my vocation.My laptop is not with me.I'll attach ASAP.

mokusolo, 2017-01-03:
https://drive.google.com/open?id=0B8Q7ChB6dmiaQnZDY3NFRzNlM2s

I tried several times to upload zip file but it returned "We dont support that file type".
So I have to put that in google drive. Wish that's available for you.
Thank you.

BagelOrb, 2017-01-06:
You can always add a bogous extension `.txt` to counter such a problem.

BagelOrb, 2017-01-06:
I think your viewer is wrong. I cannot find any instance of an `Fddddd` in the gcode. All your extrusions and travels should occur at speeds below 10000 mm/min.

Have you tried printing it?

closed 2017-03-08 by BagelOrb