Ultimaker/CuraEngine #425: Query - does thin wall print direction really need reversing every couple of layers?

smartavionics, 2016-12-27:
Printing a 30mm test cube with a solid base and thin walls that just need a single perimeter. I notice that the direction of print reverses every couple of layers. It even does this when Spiralize Outer Contour is enabled.

The problem though is that on the corner that the reversal happens, the print quality is reduced compared to the other 3 corners. I get some surface roughness approx 2mm from the corner (in both directions) that doesn't occur on the other 3 corners.

So the question is: why is it reversing the print direction of the perimeters and does it really need to given that it reduces the quality of the output?

Also, I thought the whole point of spiralize was that it could print the shell in one continuous extrusion rather than  as a zig-zag?

Ghostkeeper, 2016-12-27:
It shouldn't reverse. It normally doesn't either. I have a print running next to me now that just continues (with a small click at the layer change) in the same direction.

Does it also happen when you try other models?

smartavionics, 2016-12-27:
I haven't tried any other models yet in spiralize mode.

Here's the cube model I am using. I have to scale it to 40mm in Cura otherwise the walls don't get printed:

http://www.smartavionics.com/3dp/30mm-hollow-box.stl

And here's the settings:

;SETTING_3 {"global_quality": "[general]\\nversion = 2\\nname = Kossel XL (hot b
;SETTING_3 ed)\\ndefinition = custom\\n\\n[metadata]\\ntype = quality_changes\\n
;SETTING_3 quality_type = normal\\n\\n[values]\\nacceleration_enabled = False\\n
;SETTING_3 acceleration_wall = None\\nadhesion_type = skirt\\ncool_fan_full_at_h
;SETTING_3 eight = 1\\ncool_min_layer_time = 15\\ncool_min_layer_time_fan_speed_
;SETTING_3 max = 15\\ncool_min_speed = 15\\ninfill_before_walls = False\\ninfill
;SETTING_3 _line_width = 0.8\\ninfill_overlap = 25\\ninfill_pattern = zigzag\\ni
;SETTING_3 nfill_sparse_density = 25\\njerk_enabled = False\\njerk_wall = 5\\nla
;SETTING_3 yer_height = 0.2\\nlayer_height_0 = 0.25\\nline_width = 0.5\\nmagic_s
;SETTING_3 piralize = True\\nmaterial_diameter = 1.75\\nmaterial_print_temperatu
;SETTING_3 re = 205\\nretraction_amount = 3\\nretraction_combing = noskin\\nretr
;SETTING_3 action_extra_prime_amount = 0.15\\nretraction_hop = 0.5\\nretraction_
;SETTING_3 hop_enabled = True\\nretraction_min_travel = 2.5\\nretraction_prime_s
;SETTING_3 peed = 15\\nretraction_retract_speed = 50\\nretraction_speed = 30\\ns
;SETTING_3 kin_overlap = None\\nspeed_infill = 60\\nspeed_layer_0 = =speed_print
;SETTING_3  * 30 / 60\\nspeed_print = 60\\nspeed_topbottom = 40.0\\nspeed_travel
;SETTING_3  = =speed_print if magic_spiralize else 120\\nspeed_wall_0 = =speed_w
;SETTING_3 all\\nspeed_wall_x = 50.0\\nstart_layers_at_same_position = True\\nsu
;SETTING_3 pport_enable = False\\nsupport_pattern = concentric\\nsupport_xy_dist
;SETTING_3 ance = 2\\ntop_bottom_thickness = 0.6\\nwall_thickness = 1\\n\\n"}

BagelOrb, 2016-12-30:
I can imagine this happening if your model is not correct; if some faces
are oriented in the wrong way, then that whole layer might be printed in a
different direction.

Note that cura doesn't visualize the direction of Raj face in your model.

I do consider this a minor bug, but given that it is only in spiralize mode
and with certain 'bad' models, I think this is not important enough to put
in our priority list.

http://www.smartavionics.com/3dp/30mm-hollow-box.stl

smartavionics, 2016-12-31:
Understood. If I have any further problems with other models that look OK, I'll let you know.

BagelOrb, 2017-03-08:
I don't think this is a problem.

Perhaps what you were looking at is that the outer walls are printed in CCW order, while the inner walls are printed in CW order. That is intended behaviour. When there is a small gap between the inner and outer wall, the whole object becomes a C shape and the direction of the inner part of the C is the other direction as the outer part.

closed 2017-03-08 by BagelOrb

smartavionics, 2017-03-08:
Yes, this was back in the days when I didn't appreciate that spiralize really needs a solid model.

smartavionics, 2017-03-08:
BTW, does someone need to fix up the visualizer so that spiralized layers don't have a gap at the seam?

BagelOrb, 2017-03-08:
Yeah the layer view doesn't really support the diagonal moves.
We really couldn't be bothered to show it correctly, as spiralize is not a main feature of Cura and it would take quite a bit of work to visualize correctly.