Ultimaker/CuraEngine #426: error: ‘class Arcus::Socket’ has no member named ‘state’

GerogeFu, 2016-12-28:
when i try to compile CuraEngine in Centos 7 , after installing libArcus , i got these errors:

[ 14%] Building CXX object CMakeFiles/CuraEngine.dir/src/commandSocket.cpp.o
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp: In member function ‘void cura::CommandSocket::connect(const string&, int)’:
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:62:76: error: no matching function for call to ‘Arcus::Socket::registerMessageType(int, const Cura::ObjectList*)’
     d->socket->registerMessageType(1, &Cura::ObjectList::default_instance());
                                                                            ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:62:76: note: candidate is:
In file included from /usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:8:0:
/usr/local/include/Arcus/Socket.h:73:14: note: bool Arcus::Socket::registerMessageType(const google::protobuf::Message*)
         bool registerMessageType(const google::protobuf::Message* message_type);
              ^
/usr/local/include/Arcus/Socket.h:73:14: note:   candidate expects 1 argument, 2 provided
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:63:82: error: no matching function for call to ‘Arcus::Socket::registerMessageType(int, const Cura::SlicedObjectList*)’
     d->socket->registerMessageType(2, &Cura::SlicedObjectList::default_instance());
                                                                                  ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:63:82: note: candidate is:
In file included from /usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:8:0:
/usr/local/include/Arcus/Socket.h:73:14: note: bool Arcus::Socket::registerMessageType(const google::protobuf::Message*)
         bool registerMessageType(const google::protobuf::Message* message_type);
              ^
/usr/local/include/Arcus/Socket.h:73:14: note:   candidate expects 1 argument, 2 provided
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:64:74: error: no matching function for call to ‘Arcus::Socket::registerMessageType(int, const Cura::Progress*)’
     d->socket->registerMessageType(3, &Cura::Progress::default_instance());
                                                                          ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:64:74: note: candidate is:
In file included from /usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:8:0:
/usr/local/include/Arcus/Socket.h:73:14: note: bool Arcus::Socket::registerMessageType(const google::protobuf::Message*)
         bool registerMessageType(const google::protobuf::Message* message_type);
              ^
/usr/local/include/Arcus/Socket.h:73:14: note:   candidate expects 1 argument, 2 provided
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:65:76: error: no matching function for call to ‘Arcus::Socket::registerMessageType(int, const Cura::GCodeLayer*)’
     d->socket->registerMessageType(4, &Cura::GCodeLayer::default_instance());
                                                                            ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:65:76: note: candidate is:
In file included from /usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:8:0:
/usr/local/include/Arcus/Socket.h:73:14: note: bool Arcus::Socket::registerMessageType(const google::protobuf::Message*)
         bool registerMessageType(const google::protobuf::Message* message_type);
              ^
/usr/local/include/Arcus/Socket.h:73:14: note:   candidate expects 1 argument, 2 provided
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:66:81: error: no matching function for call to ‘Arcus::Socket::registerMessageType(int, const Cura::ObjectPrintTime*)’
     d->socket->registerMessageType(5, &Cura::ObjectPrintTime::default_instance());
                                                                                 ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:66:81: note: candidate is:
In file included from /usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:8:0:
/usr/local/include/Arcus/Socket.h:73:14: note: bool Arcus::Socket::registerMessageType(const google::protobuf::Message*)
         bool registerMessageType(const google::protobuf::Message* message_type);
              ^
/usr/local/include/Arcus/Socket.h:73:14: note:   candidate expects 1 argument, 2 provided
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:67:77: error: no matching function for call to ‘Arcus::Socket::registerMessageType(int, const Cura::SettingList*)’
     d->socket->registerMessageType(6, &Cura::SettingList::default_instance());
                                                                             ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:67:77: note: candidate is:
In file included from /usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:8:0:
/usr/local/include/Arcus/Socket.h:73:14: note: bool Arcus::Socket::registerMessageType(const google::protobuf::Message*)
         bool registerMessageType(const google::protobuf::Message* message_type);
              ^
/usr/local/include/Arcus/Socket.h:73:14: note:   candidate expects 1 argument, 2 provided
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:68:77: error: no matching function for call to ‘Arcus::Socket::registerMessageType(int, const Cura::GCodePrefix*)’
     d->socket->registerMessageType(7, &Cura::GCodePrefix::default_instance());
                                                                             ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:68:77: note: candidate is:
In file included from /usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:8:0:
/usr/local/include/Arcus/Socket.h:73:14: note: bool Arcus::Socket::registerMessageType(const google::protobuf::Message*)
         bool registerMessageType(const google::protobuf::Message* message_type);
              ^
/usr/local/include/Arcus/Socket.h:73:14: note:   candidate expects 1 argument, 2 provided
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:72:22: error: ‘class Arcus::Socket’ has no member named ‘state’
     while(d->socket->state() != Arcus::SocketState::Closed && d->socket->state() != Arcus::SocketState::Error)
                      ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:72:74: error: ‘class Arcus::Socket’ has no member named ‘state’
     while(d->socket->state() != Arcus::SocketState::Closed && d->socket->state() != Arcus::SocketState::Error)
                                                                          ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:100:24: error: ‘class Arcus::Socket’ has no member named ‘errorString’
         if(!d->socket->errorString().empty()) {
                        ^
/usr/local/src/CuraEngine-15.06/src/commandSocket.cpp:101:41: error: ‘class Arcus::Socket’ has no member named ‘errorString’
             logError("%s\n", d->socket->errorString().data());

Ghostkeeper, 2016-12-28:
It's breaking over variadic arguments in a lambda function. The code is valid, but your compiler doesn't seem to support it.

I suggest you try with a newer version of the compiler.

awhiemstra, 2016-12-28:
You are building on CentOS7, this means you'll need to get GCC 4.9 from the devtoolset 3 bundle: https://www.softwarecollections.org/en/scls/rhscl/devtoolset-3/

BagelOrb, 2016-12-30:
> /usr/local/src/CuraEngine-15.06/src/commandSocket.cpp💯24: error: ‘class

@awhiemstra perhaps you can elucidate?

awhiemstra, 2017-01-05:
@BagelOrb

That looks to me like trying to compile a fairly old CuraEngine against a fairly new libArcus. It is complaining that registerMessage(int, Message*) does not match with registerMessage(Message*). The int was removed a while ago since we now auto-generate type identifiers.

BagelOrb, 2017-07-23:
Outdated issue

closed 2017-07-23 by BagelOrb

Gaspounet, 2017-08-08:
I am having a similar issue, and I have no idea how to fix it.
I'm trying to build on Ubuntu Server and I get this error:

`~/CuraEngine/build$ make
Scanning dependencies of target clipper
[  1%] Building CXX object CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.o
Linking CXX static library libclipper.a
[  1%] Built target clipper
[  2%] Running C++ protocol buffer compiler on Cura.proto
Scanning dependencies of target _CuraEngine
[  4%] Building CXX object CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o
[  5%] Building CXX object CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o
In file included from /home/gaspounet/CuraEngine/src/pathPlanning/Comb.h:11:0,
                 from /home/gaspounet/CuraEngine/src/LayerPlan.h:8,
                 from /home/gaspounet/CuraEngine/src/FffGcodeWriter.h:15,
                 from /home/gaspounet/CuraEngine/src/FffProcessor.h:5,
                 from /home/gaspounet/CuraEngine/src/commandSocket.cpp:7:
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(Args ...)’:
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected ‘,’ before ‘...’ token
             [args...]()
                  ^
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected identifier before ‘...’ token
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:21: error: parameter packs not expanded with ‘...’:
             [args...]()
                     ^
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:21: note:         ‘args’
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:36:34: error: expansion pattern ‘args’ contains no argument packs
                 return new T(args...);
                                  ^
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<_Res(_ArgTypes ...)>&, Args ...)’:
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected identifier before ‘...’ token
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:24: note:         ‘args’
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:54:36: error: expansion pattern ‘args’ contains no argument packs
                 return new T(f(args...));
                                    ^
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<T*(Args ...)>&, Args ...)’:
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected identifier before ‘...’ token
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:24: note:         ‘args’
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/gaspounet/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:70:30: error: expansion pattern ‘args’ contains no argument packs
                 return f(args...);
                              ^
make[2]: *** [CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make: *** [all] Error 2`

Ghostkeeper, 2017-08-23:
Still the same answer: You need to get a newer compiler version!