Ultimaker/CuraEngine #431: Thin wall not printed (2.3.1 and 2.4 BETA)

Ando120, 2017-01-05:
Thin wall not printed (2.3.1 and 2.4 BETA)

Visible in layer view

example:
0.4 mm wall not printed with 0.4 nozzle.
Sometimes printing with some Z angle rotation and sometimes only part of wall printed)

Also I see that always printing thin walls with double lines. I don't understand why?
Example 0.408 wall is printed with 0.4 nozzle with two times.

I am using 0.4 wall for custom support structures to print PC

Ghostkeeper, 2017-01-05:
Walls are computed by taking the circumference of the layer and then computing an inset of half the line width (0.2mm in your case). If that results in no area left, it doesn't print anything any more. In other words, it is a known limitation of the slicing process at the moment. This is also why it makes two walls: It completes the circle.

Exactly 0.4mm is the edge case. Because we also "simplify" our polygons, which takes away lines that do not contribute to the area, they are currently deleted.

On most printers it is better to use a line width slightly lower than your nozzle size. This would be a workaround around the problem, but it won't prevent drawing that line double.

Ando120, 2017-01-05:
Ok
Unfortunately it don't print with any line with.
And sometimes it prints with some angle (z) when wall is part of bigger object. Cant find clear logic.

I will design better walls now.

BagelOrb, 2017-01-06:
I'm curious as to what you mean by the Z angle.

Could you perhaps share your gcode file?

Ando120, 2017-01-06:
If I rotate along Z in Cura then some angle its visible in layer view and some Z angle that same wall don't appear in layer view.
I redesigned my object and deleted old gcode file but will create it again tomorrow.

Ando120, 2017-01-06:
Here is STL file. There is 3 support walls 0.4 mm.
Open it with Cura layer view and rotate it( Z)
Some angle it print some not and with some angle only part of wall.

[test.zip](https://github.com/Ultimaker/CuraEngine/files/691040/test.zip)

BagelOrb, 2017-01-07:
The outer wall is printed on each side of the objects, so if you have a
0.4mm nozzle, the outer wall won't fit of both sides of a 0.4mm thick piece
in your model.

You should design your walls to be as thin as twice the nozzle size
(although pieces as thick as 1.5x the nozzle size will work quite well).

If you cannot redesign this part you can reduce the outer wall line width
to something like 0.3mm.

CCS86, 2017-02-14:
As we found in my other thread, leaving the machine nozzle width to 0.4mm and reducing the line widths, also adds a "wall inset" value, which "takes up" the space you tried to create. Reducing the machine nozzle width or zeroing the "outer wall inset" can help.

It would be great in these wall calculations, to add in a little "fuzzy math" to tip the rounding in our favor. When designing, it is much easier to work with nominals. With the practical accuracy of our printers, to be so strict about the width rounding doesn't help. It would definitely be preferable to get a wall that is 0.08mm wider than the model, as opposed to no wall at all.

I understand that the current algorithm needs to have an outer wall on "both sides" with the way it is calculated. It does really seem like there is an opportunity to rewrite this for better handling. It can already calculate the width of a thin section correct? If you had setting parameters which governed as follows:

-Minimum line width
-Maximum line width

You could evaluate thin sections as follows:

- Area is narrower than minimum line width: no print segments

- Area is between minimum width and maximum width: A single print segment is generated with the correct extrusion rate for the required length, along the area centerline.

- Area is between the maximum width and 2x nozzle diameter: Two print segments are generated, with a line width equal to the area width / 2, offset from the area centerline by the calculated line width / 2

Does that sound practical? Is the implementation harder than I am picturing?

Ghostkeeper, 2017-02-16:
It sounds ideal, but it's quite some work. The current calculation that you're referring to works a bit differently. It takes two lines that overlap, computes roughly how much they overlap (using a simplified reality that approaches it good enough), and then compensates one of the walls to have less extrusion in some part of the line.

You would compute this on the actual outlines (not inset yet by half a line width). Then you'd see how much they overlap and compensate one of the lines. This breaks the line into pieces, since some pieces of a straight line may overlap more than others. You'd shift this line into the correct direction (not easy!) so that its edge falls in the right place. You can no longer use insets here, since that would require you to compute an inset for every single line piece which is prohibitively expensive. This basically covers your third case then.

The second case would compensate the line to below the minimum line width. You'd then need to find the opposing line (we have these linked already by the current implementation). Then you'd need to break that line into multiple pieces with various thicknesses as well, and shift them again to align with the outer edge. Again, not easy, though you can re-use the other implementation.

For the first case you'd get the second line below the minimum width as well. Since the lines are already broken into pieces, this part is easy.

But in the end, this idea is probably very expensive to compute and also quite expensive to implement.

mokusolo, 2017-04-23:
I came across the problem with printing 0.9mm wall(0.4mm nozzel dia). I set wall_line_count to 3. It was supposed to print inset[0], simplifying or erasing inset[1], inset[2]. There is a gap between two walls printed, but I can't figure out why the gap still has travel path with no extrusion. It made my printing ugly. Setting wall_line_count to 2 can solve this problem, but I'm just curious about engine's calculation. Sorry for my inaccurate description.

Ghostkeeper, 2017-04-25:
Try disabling the Fill Gaps Between Walls feature. Maybe that's causing those travel moves. It could also just be that it wants to start a layer at a certain position. Hard to tell from your description.

mokusolo, 2017-04-26:
It works, amazing... I changed Fill Gaps Between Walls. Thanks a lot.

wawanbreton, 2024-04-05:
The described behavior is no more relevant. With Cura 5.0 we introduced a new slicing method which generates lines with variable extrusion width and greatly improves such issues. Please open a new issue if there is still some trouble.

closed 2024-04-05 by wawanbreton