Ultimaker/CuraEngine #432: 2.4 Beta: Cubic Subdivision Empty

CCS86, 2017-01-07:
I was excited to try out the new infill style, but found the infill empty after choosing cubic subdivision. I tried playing with the settings, but couldn't get a shred of infill.

mboerwinkle, 2017-01-07:
Just tested the current build using default settings for the printrbot simple. I was using the command line, and so used "-s infill_pattern=cubicsubdiv". It did what it was supposed to for me. Could you link or attach the model you were using so I can try it out?
Thanks,
Martin Boerwinkle

BagelOrb, 2017-01-07:
I think it is an OS thing.

For me it also didn't work on windows 7, but my colleague on Ubuntu had no
problem.

What OS do you guys have?

mboerwinkle, 2017-01-07:
I'm under 64 bit Linux.

CCS86, 2017-01-07:
Windows 10 here

BagelOrb, 2017-01-07:
Seems to work on bananas (osx 10.95)

mboerwinkle, 2017-01-08:
I don't have a Windows build platform right now, but Ill look into debugging it on Monday when I get back to school.
Thanks,
Martin Boerwinkle

mboerwinkle, 2017-01-10:
Haven't successfully built from source yet, but I have some useful clues nonetheless:
-The subdivision lines of the largest cube are printed. This is visible in the attached gcode.
-I'm Pretty Sure that the sub-cubes are not even being calculated as there was no change in execution time on the windows console application between very high and very low cube density and multipliers.
This leads me to believe that it is a problem with the subdivision choice, and not with the line generation.
Ill try and get it built on windows and actually debugging once I can.
[test.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/695338/test.gcode.txt)

mboerwinkle, 2017-01-13:
Still having problems building on Windows related to protobuf :( Someone else can tackle this if they feel like it; I don't want to delay the release. I will keep trying, but I can't guarantee that I will be successful in a reasonable time frame. 
Sorry,
Martin Boerwinkle

BagelOrb, 2017-01-13:
I hate that building can be such a hassle. I'm not really good with that kind of stuff. It's working on my Ubuntu right now, but I'm afraid I cannot help you much. Perhaps @sedwards2009 can help you with building on Windows?

Fixing this issue is on our priority list to fix before the release because people seem very interested in your feature. Hopefully we can find out what the problem is together.

Note that it only seems to be broken on Windows, so there must be some code in the subdivision of cubes which is platform dependant. It could also be the case that some memory is freed too early which is handled differently by the different operating systems.

BagelOrb, 2017-01-13:
Note that there's a compiler option to compile without libArcus and without protobuf. You can then test without the frontend by calling the engine directly, like so:
```
CuraEngine.exe slice -j resources\definitions\ultimaker2.def.json -s infill_pattern=cubicsubdiv -o test_ff.gcode -l C:\Users\TK\Desktop\T.stl
```

mboerwinkle, 2017-01-13:
OK, thanks for the help. I will look into the build without arcus/protobuf today. I'll ask if I have further questions.

mboerwinkle, 2017-01-16:
Got the windows build to cross-compile from linux. While comparing behavior to a linux build, I got more surprising results. The infill on a fresh clone/build today on native linux failed as well. The following stl returned the following gcode...
Can someone verify this? If it is failing on linux as well, it is not just an issue with fixing os compatibility and I need to start going backwards through github commits to find where it failed.
(using command "./CuraEngine slice -j printrbot_simple.def.json -s infill_pattern=cubicsubdiv -l sphere.stl -o output.gcode")
[output.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/707471/output.gcode.txt)
[sphere.stl.txt](https://github.com/Ultimaker/CuraEngine/files/707472/sphere.stl.txt)

BagelOrb, 2017-01-16:
I cannot reproduce your results.
The internal build of master for the Linux AppImage seems to produce the correct results.

Also the last build of the 2.4 branch (Jan 12) produces correct results on Ubuntu.

Are you sure your values for Cubic Subdivision Radius and Cubic Subdivision Shell are correct?

Mine are at 100% and 0.3mm.

mboerwinkle, 2017-01-17:
Bad settings were the problem. My settings were from an out of date ultimaker/cura.
The windows build produces gcode after a small modification to SubDivCube.cpp.
Everything I used is here: https://drive.google.com/drive/u/0/folders/0B6p5ArFix80vZU1HQXp6S0I2V1U
Included in the file share is SubDivCube.cpp. All the lines I modified are marked with a "//MB" comment. I also included a working executable. (It worked in wine at least!)

BagelOrb, 2017-01-17:
Awesome!

We should check if this works on our machines as well if we compile ourselves.

Could you maybe create a PR out of your changes?
You never know - maybe we have done some changes in the mean time which will get undone when we just copy your version over our version of SubDivCube.cpp.

mboerwinkle, 2017-01-17:
https://github.com/Ultimaker/CuraEngine/pull/439

BagelOrb, 2017-07-23:
Outdated issue.

closed 2017-07-23 by BagelOrb

Ghostkeeper, 2017-08-03:
This has been fixed in the meanwhile by https://github.com/Ultimaker/CuraEngine/commit/113202cd3466f9db05f7125d12f8b9b1b3542993, based on https://github.com/Ultimaker/CuraEngine/pull/439.