Ultimaker/CuraEngine #433: CuraEngine Building

ghamrawyk, 2017-01-08:
Hello everyone,
I am working on ubuntu 14.02 with CMake, and I am trying to build CuraEngine. I have succeeded to build protobuf, libArcus (at least I think so) without any errors. I reached the point where I am typing the command cmake .. for CuraEngine, when I got this error: 
CMake Error in CMakeLists.txt:
  Imported target "Arcus" includes non-existent path

    "/home/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done
-- Build files have been written to: /home/kg/dev/CuraEngine/build

Any suggestions on how to fix this, knowing that I am a beginner in cpp and ubuntu. Any explanation why is this error happening?

Thank you very much in advance.

thopiekar, 2017-01-08:
The easiest way to make sure all dependencies are installed might be for you to enable my PPA on your system, run `sudo apt update` and finally `sudo apt-get build-dep cura-engine`. This will install all dependencies, which were needed before to build the cura-engine package itself.

Regards

BagelOrb, 2017-01-08:
I'm (still) quite a beginner on Ubuntu as well, but I think that path is
rather common. What happens if you simply create that directory?

ghamrawyk, 2017-01-08:
Hey guys, thank you so much for responding. Well @thopiekar  I will try this first thing tomorrow and give you a feedback, if you don't mind. 
However I don't understand why is this problem occurring? Is it a problem in the Cmakelists of libArcus? I tried to trace it but I failed to find an answer. And what will exactly the commands you mentioned above do to fix it? I am sorry for asking too much but a brief explanation would be great to learn from you and be able to face such errors on my own in the future. Thanks a million. 

@BagelOrb When I create manually the include folder in the path shown in the error, I still get the same error.

awhiemstra, 2017-01-09:
The mentioned folder is what CMake thinks Arcus is installed to. You will need to verify the CMAKE_INSTALL_PREFIX for Arcus and make sure CuraEngine's CMake run can find that path.

Something else that might be happening is that it is finding the generated files located in Arcus' build directory before the installed version. You can fix that by seting Arcus_DIR correctly when running CMake for CuraEngine.

ghamrawyk, 2017-01-09:
@awhiemstra Thank you for the response. If you could, please, explain how can I verify the CMAKE_INSTALL_PREFIX on Ubuntu or how can I set the Arcus_DIR in the command line, I would be very thankful.

Cheers,

awhiemstra, 2017-01-09:
Easiest is to install curses-cmake-gui or whatever it is called on Ubuntu, then run "ccmake $build_dir". This gives you a simple UI that allows quick editing of CMake's configuration variables. When you run this with Arcus's build directory, you should see an entry `CMAKE_INSTALL_PREFIX` with a certain value, that value is what Arcus will be installed to. For `Arcus_DIR`, run ccmake with the CuraEngine build directory, then toggle "advanced mode". You should now have an entry "Arcus_DIR".

In addition, you can also set these variables when running CMake using `-DVARIABLE=VALUE`, for example `-DCMAKE_INSTALL_PREFIX=/some/local/path`. ccmake gives a bit more direct feedback though.

ghamrawyk, 2017-01-09:
Hey guys,
Thanks to you guys I succeeded until the part I want to run CuraEngine itself. I followed the instructions on github, but I am not sure what to do next. 
So I have Protobuf, libArcus successfully built. What is the next step? Any help?
Thanks in advance,
@thopiekar @BagelOrb @awhiemstra

ghamrawyk, 2017-01-09:
Thank you guys, I managed to do it. I read your comments somewhere else and it had to do with the version of Protobuf. Now I have the executable curaengine file. 

I have one last question though: what is the json file and what's its use? What is the command to run CuraEngine? is it this one: ./CuraEngine slice -j ~/prusa_i3.def.json -o output.gcode -l ~/source.stl           knowing that my stl file's name is "source" and my gcode output name would be "output"... what is the use of the word slice and the json file in the command? I read this command somewhere on stackflow. Any advice? I have the stl file ready in the CuraEngine folder. Any directions please...
@thopiekar @BagelOrb @awhiemstra @Ghostkeeper 

Thanks in advance,
Regards,

awhiemstra, 2017-01-09:
There's a more extensive help available when calling CuraEngine with --help, but it basically has two modes, "slice" and "connect". Slice tells the engine to slice an object, connect to connect to a socket. "connect" is used by the Cura UI. The JSON files are "machine definitions" that provide the list of available settings. For example, https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json is the base file and provides the entire tree of settings used in the Cura UI and used by the engine.

ghamrawyk, 2017-01-09:
Ok thank you for clarifying, however, as far as I understand, I should have that file on my pc in the CuraEngine folder, and adjust the settings in it as I want them and then run CuraEngine, right? 

How can I download this file, and what is then the command line to run CuraEngine, is the one I mentioned above correct, except that I will change the name of the JSON file? 

Cheers,
@awhiemstra

awhiemstra, 2017-01-09:
You can specify setting values with -s on the command line. The JSON file is used to determine what settings are available, not what values to use for them (except for defaults). You can download the files from the Cura repository. The command line for the rest looks ok.

ghamrawyk, 2017-01-09:
Ok thank you @awhiemstra  and thank you all for the help. I'll give it a try, and if there is a problem, i'll open another issue with a different name.

closed 2017-01-09 by ghamrawyk