Ultimaker/CuraEngine #434: Feature request - dynamic brimlines # when model is 'top heavy'

Thunderscreech, 2017-01-08:
Brims are used to bolster bed adhesion but a model with a skinny base that's tall can quickly end up exceeding the brim's support.

I propose treating brimlines as a minimum and increasing the number of certain conditions are met during analysis such as:

* Height trigger
* skinny 'neck' before a threshold percentage 

Treating is as a minimum would maintain backwards compatibility and dynamically increasing it could reduce the number of failed prints, especially by novice users and modelers.

BagelOrb, 2017-01-08:
We can't treat the brim line count as a minimum, because the placenta of
the object might depend on it. The build volume is limited such that all
brim lines are guaranteed to fall within the build plate area. I the
geometry of your print along with some setting then suddenly require extra
brim lines then those might fall outside the printable area.

Perhaps we should flip the idea around and decrease the number of brim
lines if the opposite of the conditions is met.

I don't really understand why the criteria you mention influence adhesion
requirements.
*Why would a higher object need a better bed adhesion?
*Why would a small layer followed by a relatively large layer require more
bed adhesion?

Thunderscreech, 2017-01-10:
> *Why would a higher object need a better bed adhesion?

As the object gets taller, the amount of torque needed to tip it/break it free of the adhesion surface goes down because of mechanical advantage.  Improved bed adhesion could lower the risk of a failed print caused by a tall, skinny object being 'wiggled' free of the bed.

> *Why would a small layer followed by a relatively large layer require more
bed adhesion?

It might not, I'm less confident that a wider brim would help something that has a 'skinny neck' partway up than I was a couple days ago.  I still think there's merit to considering a automatically wider-brim when an object is tall (especially if it's tall and skinny) but I can't support the other part of my suggestion re: the neck with data.

BagelOrb, 2017-01-13:
Ok, it seems like a good idea.

However, I think we currently don't have the time to fully test such a feature, given all the other stuff we are testing. I think we should do with one brim size which should work for all objects, for the moment.

I think the area of the bottom layer is much more important than the height, but composing a formula over area and height is going to be quite difficult.

Ghostkeeper, 2017-01-22:
An object that is wide and tall would require fewer brim lines than an object that is slim and tall.