Ultimaker/CuraEngine #435: Feature Request: Thicker Walls on holes

StefanKi, 2017-01-11:
Since most holes are printed too small (intentionally or unintentionally), or have blobs inside where the screws don't run smooth, I often drill them after printing. Then I can put the screws through without force and they have a perfect fit.
Since drilling takes off the material, the wall (and so the stability) gets smaller.

My idea: Print thicker walls around holes. Holes with 2 - 10mm are often used for screws or something like that and so they could maybe automatically identified and printed with thicker walls.
Perhaps it would be best to be able to define the width range where the walls for holes should be thicker.

bubnikv, 2017-01-11:
You mean adding an extra perimeter or two around the holes?

StefanKi, 2017-01-11:
Exactly.
Sometimes I even cut a thread into the holes.

BagelOrb, 2017-01-13:
This feature request seems very specific to a problem which only occurs in a small amount of cases.

You could try to solve this by using Infill Meshes.
* Design a model which will fall over the areas in which you want more strength.
* load it in Cura
* In the per-object settings tab, make visible: Infill Mesh, Wall Line Count, Infill Percentage, Top/Bottom Thickness
* Enable Infill Mesh for the model
* Set the Wall Line Count to zero
* set the infill percentage to 100% (or the Top/Bottom Thickness to 9999)
* Select the original model and your infill mesh
* Right click > Merge models

Ghostkeeper, 2017-01-22:
You could try adjusting the Horizontal Expansion setting, too. Setting it to -0.2mm makes models fit better according to our tests. We didn't want to make this the default in Cura because it is a patchwork solution to improve dimensional accuracy, which has a deeper problem; the layers are rounded and secretly overextruded.

BagelOrb, 2017-07-23:
We're not going to fix the dimensional accuracy with the feature request you propose, but will probably implement a different feature to fix it at some time.

closed 2017-07-23 by BagelOrb