Ultimaker/CuraEngine #437: Which piece of code generate this line of GCode?

ynilsiam, 2017-01-13:
From output.gcode,
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;LAYER:0
M107
G0 F420 X-1.531 Y-24.016 Z2.000
;TYPE:WALL-INNER
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;

- ;LAYER:0    generated by "gcode.writeLayerComment(layer_nr);" in cura::fffProcessor::writeGCode
- M107   generated by "gcode.writeFanCommand(fanSpeed);" in cura::fffProcessor::writeGCode
- ;TYPE:WALL-INNER   generated by "gcode.writeTypeComment(path->config->name);" in GCodePlanner::wirteGCode

- G0 F420 X-1.531 Y-24.016 Z2.000  
Which piece of code generated this line???

Thank you very much.

BagelOrb, 2017-01-14:
I suppose you are able to compile the engine yourself.

You can put a breakpoint in gcodeExport:;writeMove and see the stack on the
first move written.

What I often do is insert code like
```
If (location == Point(-1531, 24016))
    std::cerr < "asgsfag\n";
```
At some place and add a breakpoint there. (My IDE doesn't support
conditional breakpoints)

You could be looking for FffGcodeWriter::addPolygon or gcodeLayer::addTravel
This is just by thinking about it, though. I haven't looked at the code.

Does this help?

Ghostkeeper, 2017-01-22:
I suspect it's caused by `FffGcodeWriter::processNextMeshGroupCode`: https://github.com/Ultimaker/CuraEngine/blob/master/src/FffGcodeWriter.cpp#L305

BagelOrb, 2017-07-23:
[answered]

closed 2017-07-23 by BagelOrb